KLayout/klayout #1533: KLayout crashing with two consecutive calls of the same LayoutView::show_layout command

rizoschrist, 2023-11-17:
Klayout is crashing with the following error message when I execute the following command two times in a row:

pya.Application.instance().main_window().current_view().show_layout(layout, False)
pya.Application.instance().main_window().current_view().show_layout(layout, False)

[image]

ERROR: ../../../src/laybasic/laybasic/layCellTreeModel.cc,444,! mp_layout->under_construction () && ! (mp_layout->manager () && mp_layout->manager ()->transacting ())
ERROR: Exception caught: Internal error: ../../../src/laybasic/laybasic/layCellTreeModel.cc:444 ! mp_layout->under_construction () && ! (mp_layout->manager () && mp_layout->manager ()->transacting ()) was not true

klayoutmatthias, 2023-11-17:
Please check with the lastest version. 0.27.7 is very old.

klayoutmatthias, 2023-11-19:
Also reproducible on 0.28.12 and fixed.

Again, I'd appreciate if a bug report refers to the current version.

rizoschrist, 2023-11-20:
I can verify resolution in v0.28.13

closed 2023-11-20 by rizoschrist