KLayout/klayout #1538: What is the drawing mechanism and method of klayout?

libaineu2004, 2023-11-20:
What is the drawing mechanism and method of klayout?
I found that the software's drawing efficiency is very high, drawing tens of millions of lines is also very smooth, no delay. Is it drawn by the native qt QGraphView? Are you using any third-party libraries or other optimization methods?

klayoutmatthias, 2023-11-25:
No, KLayout uses a custom renderer which is based on monochrome bitmaps that are combined into a RGB image separately. Drawing on a monochrome image is very fast as drawing small objects is a matter of setting a single bit.

In addition, KLayout has multiple levels of optimizations such as using the box tree and layer specific cell bounding box information for short-cutting the drawing process. It also uses a rendering cache for cells so that drawing the same cell multiple times is a matter of a bitmap copy.

libaineu2004, 2023-11-27:
> No, KLayout uses a custom renderer which is based on monochrome bitmaps that are combined into a RGB image separately. Drawing on a monochrome image is very fast as drawing small objects is a matter [...]

Wow, you are amazing! What are the source files related to drawing?

olisnr, 2023-11-29:
how does monochrome bitmaps and antialiasing go together?

olisnr, 2023-11-29:
that means You do very optimized 4 or 9 times larger bitmaps, merge it into RGB and then do a 2D lowpass over it? its a luck GPUs are already invented :)

klayoutmatthias, 2023-11-30:
Well, basically yes. But it is a simple downscaling (pixel averaging) on a single RGB image. No lowpass involved and it does not require a GPU. I plain CPU is good enough.

Matthias

olisnr, 2023-12-01:
it looks very nice and response fast with my layouts. KLayout is very nice.