KLayout/klayout #1540: How to use a non-default location for libgit2?

wshanks, 2023-11-23:
I am having some trouble building a flatpak of the latest release of KLayout with git support. The flatpak build environment has a base image with some common build tools installed under /usr but everything else has to be built explicitly into /app (the only directory kept after the build process completes). So I built libgit2 with `/app/lib/libgit2.so` and `/app/include/git2.h`. In the build environment there are variables `LDFLAGS=-L/app/lib`, `CPLUS_INCLUDE_PATH=/app/include`, and `C_INCLUDE_PATH=/app/include`, but it seems like they don't make it through `build.sh` because the build ends up with `ld: cannot find -lgit2: No such file or directory`. Linking to Python and Ruby has worked okay because `build.sh` finds `python` and `ruby` on the path and adds the paths explicitly, but it does not do that for `libgit2`. Is there a way to point the build to /app for libgit2?

wshanks, 2023-11-23:
Sorry for the noise. I found a way to get this to work by setting the environment variable `LIBRARY_PATH=/app/lib`. I will leave the issue open for now in case you want to add more explicit support to `build.sh` for setting the path or for making use of `LDFLAGS`. I tried using `-expert` which I saw referenced `LDFLAGS` but other strange errors with it.

klayoutmatthias, 2023-11-23:
`-expert` with `$LDFLAGS` should be a way to pass any kind of parameters to the linker. 

I assume the problem arises because with `-expert` more variables are read, specifically `$CC` and `$CXX` which probably spoils your call. You need to set all these variables (this is the deal in expert mode). 

An easier hack is not to use expert mode, but to inject the parameter in the qmake call:

```
build.sh -qmake 'qmake QMAKE_LFLAGS=-L/app/lib' ...
```

BTW: I see in the log something like 

`... '--env=LDFLAGS=-L/app/lib -Wl,-z,relro,-z,now -Wl,--as-needed ' ...`

I am not sure how that is parsed, but maybe quotes around the string are useful:

`... '--env=LDFLAGS="-L/app/lib -Wl,-z,relro,-z,now -Wl,--as-needed"' ...

Matthias

closed 2023-11-24 by wshanks