KLayout/klayout #1542: KLayout crashes on exit

rizoschrist, 2023-11-23:
Tried in the latest v0.28.13:

```
pya.Application.instance().exit(0)
```

causing KLayout to crash on exit:
[image]

klayoutmatthias, 2023-11-23:
Negative. Doesn't crash for me (Ubuntu 22.04, .deb package from download page) when given in the console. Not when placed in a macro. Not with a layout loaded not without one.

I need more context. Try without a specific configuration by setting $KLAYOUT_HOME to some new folder (e.g. ~/.klayout-new). This will start with a fresh configuration. If the application still crashes it means the problem is not related to anything you are running of have installed.

Matthias

klayoutmatthias, 2023-11-23:
Another data point: the signal is an ABORT signal which means something goes wrong internally (with libc in that case). The terminal will show more details - they may be helpful.

rizoschrist, 2023-11-24:
Ok, I made the upgrade from v0.27.7 to 0.28.13 with the same arguments in the build script. I tried various things to understand what causes the error and I tried to build it with the `-noruby` option (I'm using v2.7.5-p203) and seems that I can't replicate the crash, so probably the versions are incompatible(?). Currently I do not need ruby in my projects, but if someone does may have the same issues.

PS: Also, probably irrelevant with the issue here, initially I was failing to run the build script due to `libgit2`, so now I have to use the `-nolibgit2 -noruby` to use the latest version (till now)

klayoutmatthias, 2023-11-24:
I ran some tests on a variety of Linux versions and I could reproduce some issue only on CentOS 7, which - in contrast to the other platforms - still uses Qt4.

I could not reproduce the problem anywhere else where Qt5 is used, so I assume that essentially it is Qt4 which produces this issue. I myself stopped using Qt4 for development and focus on Qt5 and Qt6.

Regarding Ruby, it works with Ruby 2.5 (CentOS 8), 2.7 (Ubuntu 20), 3.0 (Ubuntu 22) and later ones.

My stack trace on CentOS 7 is this:

```
pure virtual method called
terminate called without an active exception
addr2line: 'klayout': No such file
ERROR: Signal number: 6
Address: 0x10a
Program Version: KLayout 0.28.13 (2023-11-21 r5961eab)

Backtrace:
/usr/lib64/klayout/libklayout_lay.so.0 +0x295f3e lay::enable_signal_handler_gui(bool) [??:?]
/lib64/libpthread.so.0 +0xf630 __restore_rt [sigaction.c:?]
/lib64/libc.so.6 +0x36387 __GI_raise [:?]
/lib64/libc.so.6 +0x37a78 __GI_abort [:?]
/lib64/libstdc++.so.6 +0x60a95 __gnu_cxx::__verbose_terminate_handler() [??:?]
/lib64/libstdc++.so.6 +0x5ea06 std::rethrow_exception(std::__exception_ptr::exception_ptr) [??:?]
/lib64/libstdc++.so.6 +0x5ea33 std::terminate() [??:?]
/lib64/libstdc++.so.6 +0x5f59f __cxa_pure_virtual [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x4290f void std::vector<tl::Variant, std::allocator<tl::Variant> >::emplace_back<tl::Variant>(tl::Variant&&) [??:?]
/lib64/libpython3.6m.so.1.0 +0xe61ba _PyCFunction_FastCallDict [??:?]
/lib64/libpython3.6m.so.1.0 +0xa8c16 _PyObject_FastCallDict [??:?]
/lib64/libpython3.6m.so.1.0 +0xa9876 _PyObject_CallMethodId [??:?]
/lib64/libpython3.6m.so.1.0 +0x1d8fcc PyHash_GetFuncDef [??:?]
/lib64/libpython3.6m.so.1.0 +0x1d91f2 Py_FinalizeEx [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x3b51d pya::PythonInterpreter::~PythonInterpreter() [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x3b589 pya::PythonInterpreter::~PythonInterpreter() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x117e13 lay::ApplicationBase::shutdown() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x11fc66 lay::GuiApplication::shutdown() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x119034 lay::ApplicationBase::exit(int) [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0xd25b2 gsi::SerialArgs::check_data(gsi::ArgSpecBase const*) const [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x51a55 void std::__introsort_loop<__gnu_cxx::__normal_iterator<gsi::MethodBase const**, std::vector<gsi::MethodBase const*, std::allocator<gsi::MethodBase const*> > >, long>(__gnu_cxx::__normal_iterator<gsi::MethodBase const**, std::vector<gsi::MethodBase const*, std::allocator<gsi::MethodBase const*> > >, __gnu_cxx::__normal_iterator<gsi::MethodBase const**, std::vector<gsi::MethodBase const*, std::allocator<gsi::MethodBase const*> > >, long) [??:?]
/lib64/libpython3.6m.so.1.0 +0xe61ba _PyCFunction_FastCallDict [??:?]
/lib64/libpython3.6m.so.1.0 +0x15114f PyEval_EvalFrameEx [??:?]
/lib64/libpython3.6m.so.1.0 +0x145a87 _PyEval_EvalFrameDefault [??:?]
/lib64/libpython3.6m.so.1.0 +0x1515ed PyEval_EvalCodeEx [??:?]
/lib64/libpython3.6m.so.1.0 +0x15212b PyEval_EvalCode [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x3b963 pya::PythonInterpreter::eval_int(char const*, char const*, int, bool, int) [??:?]
/usr/lib64/klayout/libklayout_pya.so.0 +0x3bae5 pya::PythonInterpreter::eval_string_and_print(char const*, char const*, int, int) [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x16f792 lay::LogFile::~LogFile() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x16fe1e lay::LogFile::~LogFile() [??:?]
/lib64/libQtCore.so.4 +0x186952 QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) [??:?]
/lib64/libQtGui.so.4 +0x1ca0c8 QApplicationPrivate::notify_helper(QObject*, QEvent*) [??:?]
/lib64/libQtGui.so.4 +0x1d1cc6 QApplication::notify(QObject*, QEvent*) [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x12412d lay::GuiApplication::do_notify(QObject*, QEvent*) [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x12432e lay::GuiApplication::notify(QObject*, QEvent*) [??:?]
/lib64/libQtCore.so.4 +0x18687d QCoreApplication::notifyInternal(QObject*, QEvent*) [??:?]
/lib64/libQtGui.so.4 +0x26c3fa QX11Info::isCompositingManagerRunning() [??:?]
/lib64/libQtGui.so.4 +0x26c7b9 QX11Info::isCompositingManagerRunning() [??:?]
/lib64/libQtGui.so.4 +0x2478df QApplication::x11ProcessEvent(_XEvent*) [??:?]
/lib64/libQtGui.so.4 +0x26ef74 QX11Info::isCompositingManagerRunning() [??:?]
/lib64/libglib-2.0.so.0 +0x4c099 g_main_context_dispatch [??:?]
/lib64/libglib-2.0.so.0 +0x4c3f8 g_main_context_dispatch [??:?]
/lib64/libglib-2.0.so.0 +0x4c4ac g_main_context_iteration [??:?]
/lib64/libQtCore.so.4 +0x1b53ae QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [??:?]
/lib64/libQtGui.so.4 +0x26f0f6 QX11Info::isCompositingManagerRunning() [??:?]
/lib64/libQtCore.so.4 +0x18530f QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) [??:?]
/lib64/libQtCore.so.4 +0x18565d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) [??:?]
/lib64/libQtCore.so.4 +0x18ad29 QCoreApplication::exec() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x11e6d8 lay::GuiApplication::exec() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x123629 lay::ApplicationBase::run() [??:?]
klayout +0x455a
/usr/lib64/klayout/libklayout_rba.so.0 +0x1fcb0
/lib64/libruby.so.2.0 +0x1637d2
/lib64/libruby.so.2.0 +0x1751be
/lib64/libruby.so.2.0 +0x168b6c
/lib64/libruby.so.2.0 +0x16dd0f
/lib64/libruby.so.2.0 +0x16fd9f rb_iseq_eval_main
/lib64/libruby.so.2.0 +0x67f3a
/lib64/libruby.so.2.0 +0x6964d ruby_exec_node
/lib64/libruby.so.2.0 +0x6b18e ruby_run_node
/usr/lib64/klayout/libklayout_rba.so.0 +0x208f0 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
klayout +0x3a20
klayout +0x2fd2
/lib64/libc.so.6 +0x22555 __libc_start_main
klayout +0x31f2

Crash log written to /root/.klayout/klayout_crash.log
```

klayoutmatthias, 2023-11-25:
Maybe you can try the patches from here: https://github.com/KLayout/klayout/tree/issue-1542

This fixed my problem on CentOS 7.

Matthias

closed 2023-12-09 by klayoutmatthias