KLayout/klayout #1548: latest klayout python release failing to install on windows

joamatab, 2023-11-27:
Some gdsfactory users are reporting that the latest klayout 0.28.13 can't be installed on windows using `pip install klayout`

https://github.com/gdsfactory/gdsfactory/issues/2328#issuecomment-1828009539

klayoutmatthias, 2023-11-27:
Hi @joamatab

Delivery of the Windows packages was delayed because I was not owning the credentials on the Azure pipeline and they had expired. I had to push the packages manually which I did yesterday.

Could you check if the problem still persists?

Thanks,

Matthias

klayoutmatthias, 2023-11-27:
BTW: PyPI deployment is a permanent troublemaker. Maybe fixing your dependency to a verified release instead of the latest one makes life somewhat easier.

Matthias

joamatab, 2023-11-27:
yes, we can pin the klayout version on gdsfactory side

I tested and it works

[image]

@tvt173

closed 2023-11-27 by joamatab