KLayout/klayout #1550: Need some help figuring out what's wrong with this marker database

donn, 2023-11-30:
I have two marker databases- one generated by KLayout (for another design) and provided by a colleague, and one that is manually converted from Magic using a script.

The one from my colleague works fine:

[image]

…but the one I generated simply refuses to show the markers:

[image]

Not really sure what's going on there.

---

I've attached a reproducible for convenience: to get this design and both marker databases, unpack it and run:

```sh
klayout -e\
    -l ./sky130A.lyp\
    ./spm.klayout.gds\
    -m ./drc.klayout.xml\
    -m ./test.xml
```

[repro.tgz](https://github.com/KLayout/klayout/files/13514044/repro.tgz)

donn, 2023-12-02:
The script is public: https://github.com/efabless/openlane2/blob/main/openlane/common/drc.py

For example, this Magic DRC report, which lists rules and bounding boxes:

```
----------------------------------------
P-diff distance to N-tap must be < 15.0um (LU.3)
----------------------------------------
17.990um 21.995um 18.265um 22.995um
20.905um 22.935um 21.575um 22.995um
18.535um 21.995um 18.795um 22.635um
```

Would become this KLayout Marker Database:

```xml
<report-database>
    <cells>
        <cell>
            <name>RAM8</name>
        </cell>
    </cells>
    <categories>
        <category>
            <name>LU.3</name>
            <description>P-diff distance to N-tap must be &lt; 15.0um (LU.3)</description>
        </category>
    </categories>
    <items>
        <item>
            <cell>RAM8</cell>
            <category>'LU.3'</category>
            <visited>false</visited>
            <multiplicity>1</multiplicity>
            <values>
                <value>box: (17.990,21.995;18.265,22.995)</value>
            </values>
        </item>
        <item>
            <cell>RAM8</cell>
            <category>'LU.3'</category>
            <visited>false</visited>
            <multiplicity>1</multiplicity>
            <values>
                <value>box: (20.905,22.935;21.575,22.995)</value>
            </values>
        </item>
        <item>
            <cell>RAM8</cell>
            <category>'LU.3'</category>
            <visited>false</visited>
            <multiplicity>1</multiplicity>
            <values>
                <value>box: (18.535,21.995;18.795,22.635)</value>
            </values>
        </item>
    </items>
</report-database>
```

(You will notice the multiplicity is wrong- I fixed it in another branch but it doesn't appear to have any bearing on this functionality)

klayoutmatthias, 2023-12-02:
Thanks for pointing to the script ... however, I still don't see the issue on my installation (0.28.13, Ubuntu 22). `drc.klayout.xml` works as expected. When I click on the error shapes, the boxes are shown and the view navigates to the location.

To further debug the issue, you could try to export the markers ("File" menu in the marker browser). If you export them to a Layout and they do not show up as shapes, something is wrong with the way they are read and not with the way they are handled in the browser.

Matthias

donn, 2024-05-20:
So I finally have an update of some kind- this issue appears to be macOS-specific.

Here I have the Linux version (via X11 forwarding) and the macOS version side-by-side. The Linux version renders the markers just fine while the Mac version… doesn't.

[image]

---

I tried the export thing but then it says this and I'm not sure how to respond exactly

[image]

donn, 2024-05-20:
I was able to work around the issue by making them polygons instead of boxes. This works fine on both OSes.

klayoutmatthias, 2024-05-20:
There is nothing wrong with boxes and MacOS should display them the same way.

The MacOS screenshot does not show the interesting part from the "info" box below the marker list. It should display the value, i.e. "box: (20.905,22.935;21.575,22.995)".

BTW: Have you tried the latest 0.29.1 version?

Matthias