KLayout/klayout #1558: GDS files cannot be opened in L-Edit

Automachef, 2023-12-05:
I used the MatlabGDSPhotonics Toolbox package in matlab to generate a waveguide diagram based on modified Euler curves, but it can't be opened in Tanner L-edit 16.0, but my teacher can open it in version 13.0, I would like to ask how to solve it
[image]

chkkbim, 2023-12-05:
[1] You can register at Siemens EDA forum at [https://eda.sw.siemens.com/en-US/](url) and post your question.
Tanner L-Edit is under Siemens EDA software.

[2]  Or you can post your question at [https://www.edaboard.com/forums/analog-integrated-circuit-ic-design-layout-and-more.39/](url).

[3] You might also try asking the author at [https://www.mathworks.com/matlabcentral/fileexchange/46827-nicolasayotte-matlabgdsphotonicstoolbox](url). His most recent response was 01 Mar 2023.

klayoutmatthias, 2023-12-05:
I do not know much about L-Edit, but I'm surprised to see newer versions cannot open the file. Usually that is the other way round.

I can make some suggestions where to look at.

1. Number of points per polygon: KLayout can write up to 8000 + some more, but GDS standard is basically limited to 4000 + some more. In the writer options you can limit the number of points
2. Same for number of points per path, but KLayout does not have an option to limit the number of points per path. You can try converting to polygon and observe 1.)
3. Allowed characters in cell names or texts. KLayout does not impose any restrictions, but some tools may have some
4. Length of cell names. KLayout does not have a real limit. GDS has a limit at 32767 characters, but some tools may impose other limits
5. Non-orthogonal arrays (observe warnings on the instance properties page)
6. Layer or datatype numbers >32767
7. Paths with short edges
8. Paths with single points

This list can be continued, but this is what comes in my mind from my experience. My suggestion is to try OASIS which is specified more strictly than GDS and does not impose limits. Or maybe you have some error message you can share or - better - the file itself.

Matthias

klayoutmatthias, 2023-12-05:
Just as a confirmation: the toolbox does not use KLayout to generate GDS - it is just recommended as a viewer, right? 

So how is this issue related to KLayout?

Matthias

Automachef, 2023-12-06:
I'm sorry, I didn't make it clear that my problem is that I can open the file with Klayout software, but after saving it as a GDS2 file, it can't be opened in L-edit, and the curve path I drew has exceeded 8000 points, is it caused by this problem?
I've also attached my files, I've just started using klayout and there are many things I'm not very proficient in, I'm very sorry if there's anything wrong!
[Waveguide01.zip](https://github.com/KLayout/klayout/files/13574940/Waveguide01.zip)

stefanottili, 2023-12-06:
```
klayout Waveguide01.gds 
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=118, record number=9, cell=gdss)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=120086, record number=29, cell=gdss)
```
You've generated a gds that is "open to interpretation".
The klayout write form gives a hint on how to address the issue, change the default of 8000 to 4000.

[image]

Give that a try with L-Tanner and let us know how it goes.

You probably also want to change some parameter in the tool that generated this gds, you shouldn't need that many points to adequately represent the polygon curves.

Automachef, 2023-12-06:
Although I still don't know why it can't be opened, I installed version 13.0 of L-edit through a virtual machine, and the file was successfully opened, and I personally feel that it shouldn't be a problem with points, just I generated a circle with 360 sides and couldn't open it in L-edit
[image]
[image]

stefanottili, 2023-12-06:
https://github.com/klayoutmatthias/dump_oas_gds2

should help to figure out what the “unexpected element” might be.

klayoutmatthias, 2023-12-06:
@Automachef Maybe you can paste the last file that created the error at position 5930?

Apart from the many points there is nothing special with the file you pasted before.

Matthias

closed 2025-08-24 by Automachef