KLayout/klayout #1560: git+https package installation on windows

sebastian-goeldi, 2023-12-06:
I have ported the [metaports-info](https://github.com/gdsfactory/metainfo-ports) package to use the new git infrastructure.

On linux (debian12, built against the 0.28.13 tag with a modified `makedeb.sh` file to support debian12) this works flawlessly, but a coworker just wrote me that he gets an error on Windows. 

I can recreate the same error when running a fresh klayout installation on a windows (11) VM

[image]

I am not sure how to further debug though. I tried running klayout_app.exe from a power shell, but got no errors of info messages (maybe started it wrong though).

I'll happily assist if there is any way I can.

Sebastian

klayoutmatthias, 2023-12-06:
I found some weird behavior: the problem is caused by "libcrypto-3-x64.dll". This library isn't missing, but it appears to be working only when installed in the original MSYS2 location.

I'll keep debugging ...

Matthias

klayoutmatthias, 2023-12-07:
It's a certificate issue. I just need to figure out how to include them in the Windows distribution.

klayoutmatthias, 2023-12-07:
Got it ... the trusted CA's are not read. 

Here is a quick workaround: unzip the attached archive and copy the file `cert.pem` (taken from MSYS2) to some place and set the 'SSL_CERT_FILE' environment variable to contain the path to this file.

[cert.zip](https://github.com/KLayout/klayout/files/13603549/cert.zip)

closed 2023-12-09 by klayoutmatthias

mzdustpho, 2024-03-04:
Encountered this issue on Klayout 0.28.17 64bit on windows 11 Pro 23H2. Your suggested work around solved the issue.
Klayout is installed as Admin.
Thank You.

joamatab, 2024-03-26:
Can we fix this, we've seen other gdsfactory uers installing packages

klayoutmatthias, 2024-03-26:
cert.pem should be included now in the installation (since 0.28.14). I just confirm again, that you can install without admin privileges and installation of packages from http+git works.

So is that a glitch or a real issue?

Matthias

sebastian-goeldi, 2024-03-26:
This seems to be a real issue. Joaquin and I saw it with a user. Klayout was on 0.28.17, but still got the ssl error about github. We didn't try to use the workaround though (we just resorted to manually install the package.

Could this be an issue with existing klayout and upgrade vs completely fresh install?

klayoutmatthias, 2024-03-26:
I tried a wipe and re-install. I am not sure if the problem can be caused by other certificates present on the system. I understand that OpenSSL uses their own certificate management, but I think you can fool it if you system has SSL_CERT_FILE set to some other place.

I will try another machine without MSYS and no SSL_CERT_FILE environment variable set.

klayoutmatthias, 2024-03-26:
No success ... klayout-0.28.17-win64-install.exe on Windows 10. Wiped installation & reboot & fresh install. No admin mode. I cleaned the config directory and there is no SSL_CERT_FILE environment variable set. "cert.pem" is installed along with the main binary.

This is a pretty clean and organization-managed system without contamination.

Running KLayout and installing "metainfo_ports 0.0.6" which uses https+git works.

If you encounter the problem, could you check whether "cert.pem" was installed and there is no "SSL_CERT_FILE" environment variable set?

Matthias

klayoutmatthias, 2024-03-26:
I just noticed that cert.pem is not included in the .zip file (checked 64bit).

Maybe you installed from there instead from the installer .exe? That might explain the problem.

Matthias

mzdustpho, 2024-03-26:
Hi Matthias,

I've used klayout-0.28.17-win64-install.exe for the installation.

klayoutmatthias, 2024-03-26:
And cert.pem is there?

klayoutmatthias, 2024-03-26:
@mzdustpho You said the workaround solved the problem. So you copied cert.pem? Which means it was not there? 

I can't reproduce the problem on my side, so you need to help. My day had 14 hours already, debugging all kind of weird technical issues, and frankly, I am not in a mood for guesswork.

Did you set the SSL_CERT_FILE variable? Does the problem appear again when you remove or rename that variable? Can you check whether you might have two installations - one in your user mode and one in admin mode (search for "klayout_app.exe" on your entire system)? Can the problem be reproduced in a clean Windows installation (i.e. VM)?

Matthias

sebastian-goeldi, 2024-03-26:
I also just tried in a completely fresh VM on windows10. I even installed Anaconda because I thought that might be the culprit. Couldn't get the issue to manifest. Our case was on Win10. But I cannot confirm whether another user might not have set the SSL_CERT_FILE variable into windows vars before. The machine was a shared VM afaik.

klayoutmatthias, 2024-04-02:
Dear all,

people discovered that the Windows .zip files lacked the cert.pem (https://www.klayout.de/forum/discussion/2484/0-29-0-version-can-not-be-launched#latest). I had a bug in the repackaging script after code signing. Honestly I did not have .zip installation on my radar, but I will add that check to my release checklist.

Maybe one of you installed KLayout through the .zip, not the the .exe installer? This may explain the missing file and the connection issue.

I fixed that for 0.29.0. 

Best regards,

Matthias