KLayout/klayout #1562: gds Analyze the problem

SouthChinaHuihuisauce, 2023-12-08:
Dear author, I would like to ask how Klayout parses GDS files. If possible, could you please tell me the specific location of the code?

SouthChinaHuihuisauce, 2023-12-11:
> That's difficult to say. There is not a single piece of code responsible for reading GDS as there is an architecture behind (parser, database).
> 
> But what is the problem you're trying to address here? Or what problem do you wish to solve?

Thank you for your reply. I am trying to parse the GDS layout in Qt using some simple methods. I am wondering if it is possible to use a third-party library to parse the GDS layout. Can you give me some suggestions? In Qt, do I use QGraphicsView, QGraphicsScene, and QGraphicsItem to draw the layout? Could you please give me some suggestions? Thank you very much.

stefanottili, 2023-12-12:
It’s usually helpful to give some background when asking questions. 
Why do you want to write your own gds reader/datastructure/viewer ? 

A simple search in GitHub shows lots of gds2 related projects.
E.g. https://github.com/MichaelCache/gds_viewer

SouthChinaHuihuisauce, 2023-12-12:
> It’s usually helpful to give some background when asking questions. Why do you want to write your own gds reader/datastructure/viewer ?
> 
> A simple search in GitHub shows lots of gds2 related projects. E.g. https://github.com/MichaelCache/gds_viewer

Thank you for your reply. The purpose is to try making a demo and learn about this knowledge. Thank you very much for providing examples. Thank you.

klayoutmatthias, 2023-12-12:
@SouthChinaHuihuisauce You may want to try Python - there are some GDS2 related modules, e.g. gdstk and also PyQt5 which is a Python port for Qt. Python is usually easier to learn and to use than C++.

Matthias

SouthChinaHuihuisauce, 2023-12-13:
> @SouthChinaHuihuisauce You may want to try Python - there are some GDS2 related modules, e.g. gdstk and also PyQt5 which is a Python port for Qt. Python is usually easier to learn and to use than C++.
> 
> Matthias

Thank you for your reply. Yes, currently I am also using gdstk. I will try using gdstk in Python. I should first familiarize myself with the file structure of gds2. Thank you again for your suggestion, and I will try it out.