KLayout/klayout #1563: Duplicated Tabs in the Macro Development Tool

Kazzz-S, 2023-12-08:
Some tabs are duplicated in the Macro Development tool dialog.

**KLayout Version 0.28.12**
[image]
[image]
[image]

----

**KLayout Version 0.28.13**
[image]
[image]
[image]

klayoutmatthias, 2023-12-08:
Your right. Strange I did not notice.

There does not seem to be a negative effect, but it's ugly. I'll fix that.

Thanks,

Matthias

Kazzz-S, 2023-12-08:
I have confirmed the fix; thank you very much :-)

[image]

closed 2023-12-09 by klayoutmatthias