KLayout/klayout #1565: quit() raises an error in KLayout Python Console (0.28.14)

Kazzz-S, 2023-12-11:
I'm making `KLayout 0.28.14 DMGs` for MacOS as usual.

`HW*.dmg` packages for MacOS include the `start-console.py` script which starts the KLayout executable as a Python console interpreter. In the Mac environment, this is used to install various Python modules with the `pip` command.
If the console terminal is terminated with the `quit()` command, an error occurs
```
ERROR: Unexpected text after numeric value: '...None'.
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```
which was not present in 0.28.13 (and earlier).

----

After installing "HW-klayout-0.28.14-macOS-Monterey-1-qt5Brew-RsysPhb311.dmg":

```
MacBookPro2{sekigawa} MacOS (1)% ./start-console.py
Python 3.11.6 (main, Nov  2 2023, 04:51:19) [Clang 14.0.0 (clang-1400.0.29.202)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
(KLayout Python Console)
>>> import pip
>>> pip.main( ['list'] )
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Package         Version
--------------- ------------
contourpy       1.2.0
cycler          0.12.1
fonttools       4.46.0
kiwisolver      1.4.5
matplotlib      3.8.2
numpy           1.26.2
packaging       23.2
pandas          2.1.4
Pillow          10.1.0
pip             23.3.1
pyparsing       3.1.1
python-dateutil 2.8.2
pytz            2023.3.post1
scipy           1.11.4
setuptools      69.0.2
six             1.16.0
tzdata          2023.3
wheel           0.42.0
0
>>> quit()
ERROR: Unexpected text after numeric value: '...None'
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

The pip commands ran successfully. **So the problem is not serious.**
Also note that `sys.exit(0)` does not raise any error flags, as shown below.

```
MacBookPro2{sekigawa} MacOS (2)% ./start-console.py
Python 3.11.6 (main, Nov  2 2023, 04:51:19) [Clang 14.0.0 (clang-1400.0.29.202)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
(KLayout Python Console)
>>> sys.exit(0)
MacBookPro2{sekigawa} MacOS (3)%
```

The same symptoms are observed in "HW-klayout-0.28.14-macOS-[Ventura|Sonoma(yet to make)]-1-qt5Brew-RsysPhb311.dmg".

Kazzz-S, 2023-12-11:
I made two Linux versions (0.28.13 and 0.28.14) of `start-console.py` and compared them.

**start-console-02813.py**
```
#!/home/sekigawa/GitWork/klayout/safe02813/qt5.bin.linux-64-gcc-release/klayout-nostyle.sh -b -r
print( "KLayout version = 0.28.13 on Linux" )
import readline
import code
import sys
import os
pwd = os.getcwd()
sys.path.append(pwd)

variables = globals().copy()
variables.update(locals())
shell = code.InteractiveConsole(variables)
cprt = 'Type "help", "copyright", "credits" or "license" for more information.'
banner = "Python %s on %s\n%s\n(%s)" % (sys.version, sys.platform,
    cprt, "KLayout Python Console")
exit_msg = 'now exiting %s...' % "KLayout Python Console"
shell.interact(banner, exit_msg)
```
[image]

----

**start-console-02814.py**
```
#!/home/sekigawa/GitWork/klayout/safe02814/qt5.bin.linux-64-gcc-release/klayout-nostyle.sh -b -r
print( "KLayout version = 0.28.14 on Linux" )
import readline
import code
import sys
import os
pwd = os.getcwd()
sys.path.append(pwd)

variables = globals().copy()
variables.update(locals())
shell = code.InteractiveConsole(variables)
cprt = 'Type "help", "copyright", "credits" or "license" for more information.'
banner = "Python %s on %s\n%s\n(%s)" % (sys.version, sys.platform,
    cprt, "KLayout Python Console")
exit_msg = 'now exiting %s...' % "KLayout Python Console"
shell.interact(banner, exit_msg)
```
[image]

The problem reproduces in Linux, too.

klayoutmatthias, 2023-12-11:
Hello @Kazzz-S,

thanks as always for your efforts.

I can reproduce the issue and it is a side effect of this enhancement: https://github.com/KLayout/klayout/issues/1535. Now the SystemExit exception is taken as a string and as `quit()` raises a "None"-valued SystemExit (which I was not aware of), this secondary exception is raised.

It is easy to fix.

I have a second slightly annoying bug (https://github.com/KLayout/klayout/issues/1566), so I am considering creating a 0.28.15 version. 

What do you think?

Matthias

Kazzz-S, 2023-12-11:
Hello @klayoutmatthias,

Thanks for the quick investigation.
The error message is red and annoying, but the actual severity is low. So please include a fix in the next maintenance release ( 0.28.15).

Meanwhile, I am facing build issues with Sonoma :-(
1) After upgrading from 14.1.1 to 14.1.2, the OS itself crashed during the nightly build. I had a bad morning. This might be a problem with the virtual machine (VMware Fusion).
2) Strangely, the #1493 problem seems to reproduce despite the patch, but it happens and doesn't happen.

I'll revert to 14.1.1 and try again.

Best regards,
Kazzz-S

Kazzz-S, 2023-12-12:
> 2\. Strangely, the [qt 5.15.10 needs to be patched to build on MacOS with CommandLineTools 15 #1493](https://github.com/KLayout/klayout/issues/1493) problem seems to reproduce despite the patch, but it happens and doesn't happen.

I got the reasons.
Since there have been recent OS upgrades (Monterey ==> 12.7.2; Ventura ==> 13.6.3; Sonoma ==> 14.2) and XCode upgrades, I will rebuild ALL of them, merging the `issue-1565` branch.

Kazzz-S

Kazzz-S, 2023-12-13:
Hello @klayoutmatthias,

I've confirmed the fix.

[image]

Thanks again for the quick fix.
Kazzz-S

commit 9525cfd1 (2023-12-14): Fixed #1565 (quit() raises an error in KLayout Python Console)
closed 2023-12-14 by klayoutmatthias