KLayout/klayout #1569: Make (selected) layer toggle layer key-bindable

3e4, 2023-12-12:
Hi,

after using klayout a lot to look at complex layouts, I have some usability ideas I want to share and would be happy if they are considered for implementation.

- Double-clicking a layer will toggle visibility. This is not key-bindable. Make it key-bindable.
- Have a new layer visibility option "toggle selected", which will read current state for each selected layer/group and invert it. Make it key-bindable.

Thank you!

klayoutmatthias, 2023-12-12:
This is worth a coffee - and next time not the one that is paid from my own bonus points!

klayoutmatthias, 2023-12-14:
Added my favorite other feature: "visibility follows selection"

Kazzz-S, 2023-12-15:
Hello @klayoutmatthias,

I like the **visibility follows selection** feature! It's like the Bookmarks feature.
[image]
[ An MP4 movie: T1569.mp4.zip](https://github.com/KLayout/klayout/files/13682278/T1569.mp4.zip)

* The first half part of the movie: **UP/DOWN key** only
* The second half part of the movie: **Shift+UP/DOWN key**

Best regards,
Kazzz-S

closed 2023-12-16 by klayoutmatthias

klayoutmatthias, 2023-12-16:
I honestly don't know why I did not come up with the "visibility follows selection" feature earlier. 

But thanks go to @3e4 who keeps bugging me about such features ;)

Matthias