KLayout/klayout #1571: Marker Database Browser doesn't keep Cell/Category tab size persistent

stefanottili, 2023-12-12:
This is an enhancement request.

[image]

It would be helpful if

a) the Cell/Category would be resized to display the full Cell/Category text  ..
and/or
b) the Cell/Category tab size would be persistent.
Currently it has to be resized every time klayout gets started.

stefanottili, 2024-02-29:
If this get's addressed, it would be very helpful if the "By Category" would be either always expanded or persistent.
This will save yet another mouse click that is currently needed for every time errors are viewed.

klayoutmatthias, 2024-03-23:
I just tried to reproduce the problem, but at least on Ubuntu-22 with Qt 5.15 the browser window's dimensions and splitter section sizes are persisted (that is part b).

This is what I do:

* I load some layout
* A create a marker database with "Shapes to markers"
* I adjust the dimensions of the marker browser window's elements
* I exit KLayout

When I reopen and do "Shapes to markers" again, the marker browser window opens with the previous geometry.

klayoutmatthias, 2024-03-23:
Looks like the problem is Windows specific. On Windows, the column widths are not persisted :(

stefanottili, 2024-03-24:
I read rdb via command line, but that shouldn’t matter.So it might be a Mac only issue.

klayoutmatthias, 2024-03-29:
I need a little more time to debug the problem. It is likely that this is a Qt issue or I am using "restoreState" incorrectly. 

I will schedule that for 0.29.1.

Matthias

klayoutmatthias, 2024-04-29:
The closest way I came to a solution is by explicitly setting the section resize mode to "interactive". I assume that on some OS (like Windows), the resize mode defaults to "automatic", hence this problem.

klayoutmatthias, 2024-06-08:
Sorry, need to shift milestone again. No clue yet :(

stefanottili, 2024-07-06:
I build klayout with qt 6.7 on MacOS and the tab sizes are persistent now !!! Yay.

It still would save me one click every time when the "Category" would remember that it was expanded last time.

closed 2024-07-06 by stefanottili