KLayout/klayout #1576: Open dialogs share the same history

mole99, 2023-12-17:
First of all, thank you for your work on KLayout!

This is just an enhancement request for convenience.

The open dialogs seem to share the same history, at least on Linux.

For example, if you open a layout with `File -> Open`, KLayout will remember the last used directory.
If you then load layer properties with `File -> Load Layer Properties` the file explorer is opened where `File -> Open` left off.
If you now open a layout again with `File -> Open`, the file explorer is opened where `File -> Load Layer Properties` left off.

It would be great if the different open dialogs would not share their history.

klayoutmatthias, 2023-12-19:
@mole99 Thanks for this report, but I can't confirm that. I am on Ubuntu 22.

Here is what I do:

1. I start KLayout from my home directory
2. I use File/Open and navigate to some location, open a layout file from there
3. I use File/Load Layer Properties. It starts from my home folder. I navigate to some other place and load a .lyp there.
4. When I use File/Open it starts from where I left it in 2.
5. When I use File/Load Layer Properties again, it starts from where I left it in 3.

On Windows, the file browser brings it's own "smartness" which interferes with KLayout's attempts to store a history per file domain. But I have not seen that on Linux. Maybe it is a problem with the File Browser that comes with your Window Manager or a specific Qt style which comes with a special File Browser?

Matthias

mole99, 2023-12-20:
Sorry! I forgot to mention that I installed KLayout via Flatpak.

When I install KLayout the normal way, the open dialogs don't share their history. So it seems this is a Flatpak specific issue, and you may not even be able to change the behavior. Thanks anyway for taking a look at it :)

I have updated the title of the issue report.

Leo

klayoutmatthias, 2023-12-22:
Thanks for mentioning this.

I think this is not necessarily a problem of flatpak, but more a Qt issue: Qt has several ways of integrating file dialogs and comes with its own implementation as fallback. Specifically if the file dialog is borrowed from the system, they come with their own smartness and override my attempts to point them to the last location that was used for a specific purpose.

On my Ubuntu 22 the GTK3/Gnome file dialog respects the given paths, but other implementations may not do so.

Matthias

klayoutmatthias, 2024-03-23:
I'll close this ticket as I don't think there is much I can do.

closed 2024-03-23 by klayoutmatthias