KLayout/klayout #1583: Add possibility to customize main window title

EugeneZelenko, 2024-01-04:
Hi, Matthias!

It'll be great to have possibility to customize main window title using command line option or environment variable. Motivation for this feature came from application that uses KLayout as viewer and could launch multiple KLayout sessions, so it's hard to distinguish between different applications in task bar.

klayoutmatthias, 2024-01-05:
I'd propose a new attribute of the `MainWindow` class which allows configuring the window title.

Currently, you can configure the layout view names differently by using something like:

```
pya.LayoutView.current().title = "ABC"
```

this will show up in the main window title like this:

[image]

So that is a way to customize at least the back part of the title.

Matthias

EugeneZelenko, 2024-01-05:
Hi, Matthias!

Sure, this will work! Title could be set in startup script (`-rm`) or synchronized in more complex scenarios like socket/pipe monitoring.

closed 2024-03-29 by klayoutmatthias

EugeneZelenko, 2024-03-29:
Hi, Matthias!

Thank you for enhancement!