KLayout/klayout #1584: Text display issues

Trabbish, 2024-01-04:
Klayout version is 0.28.14, and in the setup window a part of the word has a large and small letter spacing. The same thing happens in the main screen under the Cells Views.
[image]
[image]

Kazzz-S, 2024-01-04:
In my environment...
[image]
[image]
[image]

Trabbish, 2024-01-05:
Thank you, this issue is caused by win11

closed 2024-01-05 by Trabbish