KLayout/klayout #1586: DCplxTrans conversion accuracy

s910324, 2024-01-05:
Hi Matthias,

I found while doing `DCplxTrans` to `ICplxTrans` conversion the `disp` loose alot of accuracy, basically the value kind of being rounded to integer first before being convert to dbu unit.

Unlike DTrans to Trans, `disp` was first divided by dbu then being convert to int, we only loose the accuracy due to the grid size, after DCplxTrans --> ICplxTrans conversion 12.34567 um becomes 12.00 um 

Currently I always extract the `disp` part and convert the value separately to a new `ICplxTrans`, I was wondering whether there's a suggested way to make this type of conversion?

``` python
import pya
dbu = 0.001
t1  = pya.DCplxTrans(1.0, 45.0, False , 12.345678, 20.000000)
t2  = pya.DTrans(12.345678, 20.000000)

print (f"T1 to dbu type: {dcplxtrans.to_itrans(dbu)}")
print (f"T2 to dbu type: {dtrans.to_itype(dbu)}")

```

```
>>> T1 to dbu type: r45 *1 12000,20000
>>> T2 to dbu type: r0 12346,20000
````

klayoutmatthias, 2024-01-05:
Hi @s910324,

thanks for reporting this issue. It is actually easy to fix, but I just released 0.28.15, so it did not go there for now.

As a workaround I'd suggest the canonical way, i.e. combining three transformations: one going into micrometer space, then the micrometer-space t1 and a last transformation going back to DBU space:

```
to_um = pya.CplxTrans(dbu)
from_um = to_um.inverted()
t1_itrans = from_um * t1 * to_um

print (f"(correct) T1 to dbu type: {t1_itrans}")
```

This gives you:

```
(correct) T1 to dbu type: r45 *1 12345.678,20000
```

Side note: You see in that case, that the displacement of the `ICplxTrans` is even non-integer. Although `ICplxTrans` transforms integer coordinates to integer ones, internally it uses floating-point representations. This allows implementing "precise" scaling and rotation with rounding only happening on delivery of the transformed coordinates (late rounding). Consequently, the displacement is floating-point too. The effect is that you should get the same results, regardless whether you first transform the coordinates to micrometer and transform it with a micrometer-unit transformation or whether you directly apply the corresponding `ICplxTrans`.

I also noted the "to_xxx" names are messed up for `ICplxTrans` and `VCplxTrans` (e.g. `ICplxTrans#to_itrans` actually should be `to_dtrans`). I guess I will clean this up too.

klayoutmatthias, 2024-04-04:
0.29 is released now :)

closed 2024-04-04 by klayoutmatthias