KLayout/klayout #1589: strm2*.exe files do not run in the Windows Command Shell and Power Shell

Kazzz-S, 2024-01-06:
Hello @klayoutmatthias,

Thanks for fixing issue #1578.
Unfortunately, these Buddy tools, including those built locally from source, do not run in the Windows Command Shell and Power Shell.
However, **`they can run in a MinGW64 terminal`**.

I guess that they were built as GUI applications, as shown below.
[image]

They should have been built as console applications, like
[image]

A change I made many years ago may be related to this issue.
[image]

Best regards,
Kazzz-S

klayoutmatthias, 2024-01-06:
Hi @Kazzz-S,

you're right, but I am not sure if that actually is bad.

From what I see, the strm* utilities basically work also from cmd.exe, but the output is not visible. You can however, redirect the output to a file or pipe, like this:

```
strm2gds -h | more
```

I recall that console applications open their own console window when run from outside the shell - which is kind of ugly. I am sure it is possible to suppress this behavior in some way, but my experience dates back to a time when this wasn't the case. 

Matthias

Kazzz-S, 2024-01-08:
Hi @klayoutmatthias,

I have confirmed that the ` | more` pipe works. Thanks for the clarification.

----

I also tested my idea above, and it looks OK, as shown in the five pictures below.
Please note that I have tested `$ ./scripts/deploy-win-mingw.sh  -64` only.

**[Chages]**
[image]

**[GUI ---> console]**
[image]

**[In the MinGW64 Terminal]**
[image]

**[In the Windows Power Shell]**
[image]

**[In the Windows Command Shell]**
[image]

Regards,
Kazzz-S

klayoutmatthias, 2024-01-09:
Hi @Kazzz-S,

I verified that "CONFIG += console" actually fixes the problem.

However, if I start the "strm*" tools as a subprocess from a GUI application, for example from KLayout's IDE console:

```
> import subprocess
# Change to your path:
> inst = "c:\\msys64\\home\\matth\\klayout\\build-release-win64"
> print(subprocess.check_output([inst + "\\strmrun", "--version"]))
```

then I see a console window flashing open briefly. This is the behavior I remembered and which I consider somewhat ugly. Someone running the strmxxx tools as subprocesses from his application - for example to do layout processing in the background - cannot stop these windows from popping up.

With a GUI application this isn't happening. In the past, I have not found a way to suppress this behavior. 

Matthias

Kazzz-S, 2024-01-09:
Hi @klayoutmatthias,

To be honest, I had never imagined a use case like the example above. If you have that kind of background, I think building them as GUI applications is fine. I think it would be nice if there was a cautionary document somewhere (if it doesn't exist) about how to use a PIPE when using a strmxxx alone in a command shell.

Kazzz-S

Kazzz-S, 2024-01-09:
Question resolved and closed.

closed 2024-01-09 by Kazzz-S