KLayout/klayout #1593: Certificate no longer for klayout.de

emipa606, 2024-01-17:
Seems the certificate has issues for the klayout.de-page since today.
`NET::ERR_CERT_AUTHORITY_INVALID`
Is it expired or is there a larger issue here?

klayoutmatthias, 2024-01-18:
Thanks a lot for this notification. Once a year I have to manually renew the certificate and I tend to forget that. 
But it is fixed now.

Best regards,

Matthias

closed 2024-01-18 by klayoutmatthias

sebastian-goeldi, 2024-01-19:
@klayoutmatthias may I suggest then to use an automatic setup which renews the certificate(s) automatically?

I have had very good experiences with certbot from letsencrypt.org. I am personally using gandi.net and then the dns plugin for it https://github.com/obynio/certbot-plugin-gandi . This has been rock-solid for me for the last 5+ years (there was only a problem once when the api key changed on gandi's side, but that was more a me problem than certbot).

Looking at the whois entry, I assume you are using godaddy? There is also a dns plugin for it. But assuming the server whoch should obtain the certificate has an open port 443 (and maybe 80), no dns plugin is necessary and just certbot is sufficient.

If this is something you would be interested in, I would be more than happy to help set it up.

klayoutmatthias, 2024-01-19:
I'm using LetsEncrypt already for klayout.org, but klayout.de is hosted differently with a number of restrictions for historical reasons. As the Forum is hosted here too, moving it somewhere else will be some effort I did not spend so far.