KLayout/klayout #1599: Compile Errors with Homebrew Qt5.15.12

Kazzz-S, 2024-01-24:
This is not a bug in KLayout but an environmental problem.
I first found this problem on **(Intel) macOS Monterey** while building [a recent commit](https://github.com/KLayout/klayout/commit/f1f92e5a0ca7b8aa85295bc69e4827ebc29a114f), but it is not macOS specific.
Under certain conditions, the problem reproduces on Linux as well.

I hope this doesn't happen on the many Linuxes for which the binaries are provided!

**1. The first problem on Monterey with `Homebrew Qt5.15.12`.**
```
:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -std=c++11 -fvisibility=hidden -O2 -std=gnu++11  -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -mmacosx-version-min=10.13 -Wall -Wextra -pedantic -Woverloaded-virtual -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_GIT2 -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=30300 -DHAVE_QT -DHAVE_QT_UITOOLS -DHAVE_QT_NETWORK -DHAVE_QT_SQL -DHAVE_QT_SVG -DHAVE_QT_PRINTSUPPORT -DHAVE_QT_MULTIMEDIA -DHAVE_QT_DESIGNER -DHAVE_QT_XML -DKLAYOUT_VERSION=0.28.15 -DKLAYOUT_VERSION_REV=a67a28c8f -DKLAYOUT_VERSION_DATE=2024-01-22 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DKLAYOUT_TINY_VERSION=15 -DMAKE_GSI_QTNETWORK_LIBRARY -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_DESIGNER_LIB -DQT_UIPLUGIN_LIB -DQT_MULTIMEDIAWIDGETS_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_XML_LIB -DQT_XMLPATTERNS_LIB -DQT_NETWORK_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I../../../../src/gsiqt/qt5/QtNetwork -I/Users/sekigawa/GitWork/klayout/qt5Brew.build.macos-Monterey-release-Rhb33Phb311/gsiqt/qt5/QtNetwork -I../../../../src/tl/tl -I../../../../src/gsi/gsi -I../../../../src/gsiqt/qtbasic -I/usr/local/Cellar/qt@5/5.15.12/lib/QtPrintSupport.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtDesigner.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtUiPlugin.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtMultimediaWidgets.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtSvg.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/include -I/usr/local/Cellar/qt@5/5.15.12/include/QtUiTools -I/usr/local/Cellar/qt@5/5.15.12/lib/QtWidgets.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtMultimedia.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtGui.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtXml.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtXmlPatterns.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtNetwork.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtSql.framework/Headers -I/usr/local/Cellar/qt@5/5.15.12/lib/QtCore.framework/Headers -I/Users/sekigawa/GitWork/klayout/qt5Brew.build.macos-Monterey-release-Rhb33Phb311/gsiqt/qt5/QtNetwork -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/AGL.framework/Headers -I/usr/local/include -I/usr/local/Cellar/qt@5/5.15.12/mkspecs/macx-clang -F/usr/local/Cellar/qt@5/5.15.12/lib -o gsiDeclQDtls.o ../../../../src/gsiqt/qt5/QtNetwork/gsiDeclQDtls.cc
In file included from ../../../../src/gsiqt/qt5/QtNetwork/gsiDeclQDtls.cc:30:
In file included from /usr/local/Cellar/qt@5/5.15.12/lib/QtNetwork.framework/Headers/QDtls:1:
/usr/local/Cellar/qt@5/5.15.12/lib/QtNetwork.framework/Headers/qdtls.h:52:1: error: static_assert failed due to requirement 'bool(-1 == 1)' "Required feature dtls for file /usr/local/Cellar/qt@5/5.15.12/lib/QtNetwork.framework/Headers/qdtls.h not available."
QT_REQUIRE_CONFIG(dtls);
^~~~~~~~~~~~~~~~~~~~~~~
:
```

ChatGPT says:
```
This error message indicates that the required feature (dtls) is not available in the Qt build configuration.
When Qt is compiled, certain features, including dtls, need to be enabled.
If dtls support is disabled, attempting to compile files that depend on it, such as qdtls.h, will result in this error.
```

Unfortunately, the binary distribution does not provide detailed information about what features were enabled when it was built.
In particular, the **config.summary** file is usually not included in the binary distribution.

Kazzz-S, 2024-01-24:
**2. Test on Linux with `Qt5.15.7 built from its source`**
The problem is reproduced as shown below.
```
:
/usr/local/bin/clang++ -c -I/usr/local/lib/clang/17/include -I/usr/local/include -I/usr/include/x86_64-linux-gnu -I/usr/include -std=c++11 -fvisibility=hidden -Wall -Wextra -pedantic -Woverloaded-virtual -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -D_REENTRANT -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_GIT2 -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=30103 -DHAVE_QT -DHAVE_QT_UITOOLS -DHAVE_QT_NETWORK -DHAVE_QT_SQL -DHAVE_QT_SVG -DHAVE_QT_PRINTSUPPORT -DHAVE_QT_MULTIMEDIA -DHAVE_QT_DESIGNER -DHAVE_QT_XML -DKLAYOUT_VERSION=0.28.15 -DKLAYOUT_VERSION_REV=a67a28c8f -DKLAYOUT_VERSION_DATE=2024-01-23 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DKLAYOUT_TINY_VERSION=15 -DMAKE_GSI_QTXML_LIBRARY -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_DESIGNER_LIB -DQT_UIPLUGIN_LIB -DQT_MULTIMEDIAWIDGETS_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_XML_LIB -DQT_XMLPATTERNS_LIB -DQT_NETWORK_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I../../../../src/gsiqt/qt5/QtXml -I/home/sekigawa/GitWork/klayout/Clang-qt5.build.linux-64-gcc-release/gsiqt/qt5/QtXml -I../../../../src/tl/tl -I../../../../src/gsi/gsi -I../../../../src/gsiqt/qtbasic -I/usr/local/Qt-5.15.7/include -I/usr/local/Qt-5.15.7/include/QtPrintSupport -I/usr/local/Qt-5.15.7/include/QtDesigner -I/usr/local/Qt-5.15.7/include/QtUiPlugin -I/usr/local/Qt-5.15.7/include/QtMultimediaWidgets -I/usr/local/Qt-5.15.7/include/QtSvg -I/usr/local/Qt-5.15.7/include/QtUiTools -I/usr/local/Qt-5.15.7/include/QtWidgets -I/usr/local/Qt-5.15.7/include/QtMultimedia -I/usr/local/Qt-5.15.7/include/QtGui -I/usr/local/Qt-5.15.7/include/QtXml -I/usr/local/Qt-5.15.7/include/QtXmlPatterns -I/usr/local/Qt-5.15.7/include/QtNetwork -I/usr/local/Qt-5.15.7/include/QtSql -I/usr/local/Qt-5.15.7/include/QtCore -I/home/sekigawa/GitWork/klayout/Clang-qt5.build.linux-64-gcc-release/gsiqt/qt5/QtXml -I/usr/local/Qt-5.15.7/mkspecs/linux-g++ -o gsiDeclQDomNotation.o ../../../../src/gsiqt/qt5/QtXml/gsiDeclQDomNotation.cc
In file included from ../../../../src/gsiqt/qt5/QtNetwork/gsiDeclQDtls.cc:30:
In file included from /usr/local/Qt-5.15.7/include/QtNetwork/QDtls:1:
/usr/local/Qt-5.15.7/include/QtNetwork/qdtls.h:52:1: error: static assertion failed due to requirement 'bool(-1 == 1)': Required feature dtls for file /usr/local/Qt-5.15.7/include/QtNetwork/qdtls.h not available.
   52 | QT_REQUIRE_CONFIG(dtls);
      | ^~~~~~~~~~~~~~~~~~~~~~~
:
```

The image below clearly shows that the DTLS feature is not enabled.
[image]

Kazzz-S, 2024-01-24:
**3. Tests on Monterey with `MacPorts Qt5.15.12` and `Anaconda3 Qt5.15.2`**
I was able to successfully build the commit in question in both of these environments.
So let's look at the corresponding symbols in the three binary libraries

**(1) Homebrew Qt=5.15.12**

```
MacBookPro2{sekigawa} 5 (1)% pwd
/usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5

MacBookPro2{sekigawa} 5 (2)% nm QtNetwork | grep Dtls | c++filt
(!empty!) <====
MacBookPro2{sekigawa} 5 (3)%
```

**(2) MacPorts Qt=5.15.12**
```
MacBookPro2{sekigawa} 5 (1)% pwd
/opt/local/libexec/qt5/lib/QtNetwork.framework/Versions/5

MacBookPro2{sekigawa} 5 (2)% nm QtNetwork | grep Dtls | c++filt
0000000000087928 T QSslConfiguration::defaultDtlsConfiguration()
0000000000087940 T QSslConfiguration::setDefaultDtlsConfiguration(QSslConfiguration const&)
000000000008790e T QSslConfiguration::setDtlsCookieVerificationEnabled(bool)
00000000000a5dae T QDtlsClientVerifier::qt_metacall(QMetaObject::Call, int, void**)
00000000000a5d6e T QDtlsClientVerifier::qt_metacast(char const*)
000000000008f8b8 T QDtlsClientVerifier::verifyClient(QUdpSocket*, QByteArray const&, QHostAddress const&, unsigned short)
00000000000c3620 S QDtlsClientVerifier::staticMetaObject
000000000008f7a0 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters(QCryptographicHash::Algorithm, QByteArray const&)
000000000008f774 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters()
000000000008f78c T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters(QCryptographicHash::Algorithm, QByteArray const&)
000000000008f75c T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters()
000000000008f872 T QDtlsClientVerifier::setCookieGeneratorParameters(QDtlsClientVerifier::GeneratorParameters const&)
000000000008f838 T QDtlsClientVerifier::QDtlsClientVerifier(QObject*)
000000000008f7b4 T QDtlsClientVerifier::QDtlsClientVerifier(QObject*)
000000000008f856 T QDtlsClientVerifier::~QDtlsClientVerifier()
000000000008f84c T QDtlsClientVerifier::~QDtlsClientVerifier()
000000000008f842 T QDtlsClientVerifier::~QDtlsClientVerifier()
000000000008fd0c T QDtls::setMtuHint(unsigned short)
000000000008fd9a T QDtls::doHandshake(QUdpSocket*, QByteArray const&)
00000000000a5e2c T QDtls::pskRequired(QSslPreSharedKeyAuthenticator*)
00000000000a5ef4 T QDtls::qt_metacall(QMetaObject::Call, int, void**)
00000000000a5eb4 T QDtls::qt_metacast(char const*)
000000000008ff8e T QDtls::handleTimeout(QUdpSocket*)
0000000000090082 T QDtls::abortHandshake(QUdpSocket*)
000000000008fe10 T QDtls::startHandshake(QUdpSocket*, QByteArray const&)
0000000000090278 T QDtls::decryptDatagram(QUdpSocket*, QByteArray const&)
000000000008fff4 T QDtls::resumeHandshake(QUdpSocket*)
00000000000a5e7e T QDtls::handshakeTimeout()
00000000000c3650 S QDtls::staticMetaObject
000000000008fef8 T QDtls::continueHandshake(QUdpSocket*, QByteArray const&)
000000000008fa7a T QDtls::setDtlsConfiguration(QSslConfiguration const&)
00000000000901ea T QDtls::writeDatagramEncrypted(QUdpSocket*, QByteArray const&)
000000000008fc2c T QDtls::setPeerVerificationName(QString const&)
000000000009036a T QDtls::ignoreVerificationErrors(QVector<QSslError> const&)
000000000008fd28 T QDtls::setCookieGeneratorParameters(QDtlsClientVerifier::GeneratorParameters const&)
000000000008fb1c T QDtls::setPeer(QHostAddress const&, unsigned short, QString const&)
0000000000090116 T QDtls::shutdown(QUdpSocket*)
000000000008fae2 T QDtls::QDtls(QSslSocket::SslMode, QObject*)
000000000008f9f8 T QDtls::QDtls(QSslSocket::SslMode, QObject*)
000000000008fb00 T QDtls::~QDtls()
000000000008faf6 T QDtls::~QDtls()
000000000008faec T QDtls::~QDtls()
00000000000a5d50 T QDtlsClientVerifier::metaObject() const
000000000008f9a6 T QDtlsClientVerifier::verifiedHello() const
000000000008f9d8 T QDtlsClientVerifier::dtlsErrorString() const
000000000008f888 T QDtlsClientVerifier::cookieGeneratorParameters() const
000000000008f9ca T QDtlsClientVerifier::dtlsError() const
00000000000a5e96 T QDtls::metaObject() const
000000000008fcaa T QDtls::peerAddress() const
[... 78 lines cut ...]
00000000000889c2 T QDtlsClientVerifier::dtlsError() const
000000000009ce86 T QDtls::metaObject() const
0000000000088c58 T QDtls::peerAddress() const
00000000000890ec T QDtls::sessionCipher() const
0000000000088cfc T QDtls::handshakeState() const
0000000000089240 T QDtls::dtlsErrorString() const
0000000000089106 T QDtls::sessionProtocol() const
0000000000088cea T QDtls::dtlsConfiguration() const
0000000000088c78 T QDtls::peerVerificationName() const
00000000000890e0 T QDtls::isConnectionEncrypted() const
0000000000089256 T QDtls::peerVerificationErrors() const
0000000000088cc4 T QDtls::cookieGeneratorParameters() const
0000000000088ca8 T QDtls::mtuHint() const
0000000000088c92 T QDtls::sslMode() const
0000000000088c6e T QDtls::peerPort() const
0000000000089236 T QDtls::dtlsError() const
00000000000ea678 S typeinfo for QDtlsClientVerifier
00000000000ea700 S typeinfo for QDtls
00000000000b16c4 S typeinfo name for QDtlsClientVerifier
00000000000b16da S typeinfo name for QDtls
00000000000ea608 S vtable for QDtlsClientVerifier
00000000000ea690 S vtable for QDtls

(base) MacBookPro2{sekigawa} lib (4)%
```

Kazzz-S, 2024-01-24:
**4. Tests on Linux Mint 20.3 with `Qt5.12.8` distributed binary**
I was also able to successfully build the commit in question in this environment.
I cannot see the symbols in the library because it is stripped.

```
MyHost{sekigawa}(1)$ pwd
/usr/lib/x86_64-linux-gnu

MyHost{sekigawa}(2)$ file libQt5Network.so.5.12.8
libQt5Network.so.5.12.8: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=18dea89e209731be37faeb284d78c0da74c3a978, stripped

```

klayoutmatthias, 2024-01-24:
Hi @Kazzz-S,

many thanks for this report. This pull request is a very big one and is supposed to form the basis of the 0.29 branch. It includes goodies such as keyword arguments for Python, Delaunay decomposition and other things. 

I am currently debugging issues on Window too which where not detected in the CI runs. So far, this branch is not fully stable. The stable branch is 0.28 and I will create another minor release from it (0.28.16), so the stability is not compromised.

Matthias

Kazzz-S, 2024-01-24:
**5. Rebuild of `Qt5.15.7` for Linux from the source code**
Following the suggestions by `ChatGPT`, I rebuilt `Qt5.15.7` with the following parameters.

```
$ sudo ./configure  -opensource  -no-strip  -release  -openssl-linked
                                                      ^^^^^^^^^^^^^^^
```
[image]

```
MyHost{sekigawa}(1)$ pwd
/usr/local/Qt-5.15.7/lib

MyHost{sekigawa}(2)$ nm libQt5Network.so.5.15.7 | grep Dtls | c++filt
000000000012c860 t q_getDefaultDtlsCiphers()
000000000012ed20 t q_setDefaultDtlsCiphers(QList<QSslCipher> const&)
00000000001853a0 r qt_meta_data_QDtls
0000000000185420 r qt_meta_stringdata_QDtls
0000000000185520 r qt_meta_data_QDtlsClientVerifier
0000000000185560 r qt_meta_stringdata_QDtlsClientVerifier
00000000001590d0 t dtlsopenssl::DtlsState::setLinkMtu(QDtlsBasePrivate*)
0000000000159b60 t dtlsopenssl::DtlsState::initCtxAndConnection(QDtlsBasePrivate*)
000000000015a240 t dtlsopenssl::DtlsState::init(QDtlsBasePrivate*, QUdpSocket*, QHostAddress const&, unsigned short, QByteArray const&)
0000000000158880 t dtlsopenssl::DtlsState::reset()
0000000000158ef0 t dtlsopenssl::DtlsState::initBIO(QDtlsBasePrivate*)
000000000015a130 t dtlsopenssl::DtlsState::initTls(QDtlsBasePrivate*)
0000000000134f60 t QDtlsBasePrivate::isDtlsProtocol(QSsl::SslProtocol)
0000000000136490 t QDtlsBasePrivate::setConfiguration(QSslConfiguration const&)
0000000000134e30 t QDtlsBasePrivate::setCookieGeneratorParameters(QCryptographicHash::Algorithm, QByteArray const&)
000000000015dd70 t QDtlsBasePrivate::~QDtlsBasePrivate()
000000000015c2a0 t QDtlsBasePrivate::~QDtlsBasePrivate()
000000000015c2a0 t QDtlsBasePrivate::~QDtlsBasePrivate()
000000000011cbb0 T QSslConfiguration::defaultDtlsConfiguration()
000000000011cc00 T QSslConfiguration::setDefaultDtlsConfiguration(QSslConfiguration const&)
000000000011d1e0 T QSslConfiguration::setDtlsCookieVerificationEnabled(bool)
0000000000168450 T QDtlsClientVerifier::qt_metacall(QMetaObject::Call, int, void**)
0000000000168390 T QDtlsClientVerifier::qt_metacast(char const*)
0000000000135050 T QDtlsClientVerifier::verifyClient(QUdpSocket*, QByteArray const&, QHostAddress const&, unsigned short)
00000000001c18a0 D QDtlsClientVerifier::staticMetaObject
0000000000168240 t QDtlsClientVerifier::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)
0000000000134f90 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters(QCryptographicHash::Algorithm, QByteArray const&)
0000000000134f70 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters()
0000000000134f90 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters(QCryptographicHash::Algorithm, QByteArray const&)
0000000000134f70 T QDtlsClientVerifier::GeneratorParameters::GeneratorParameters()
0000000000135000 T QDtlsClientVerifier::setCookieGeneratorParameters(QDtlsClientVerifier::GeneratorParameters const&)
0000000000138740 T QDtlsClientVerifier::QDtlsClientVerifier(QObject*)
0000000000138740 T QDtlsClientVerifier::QDtlsClientVerifier(QObject*)
0000000000134fe0 T QDtlsClientVerifier::~QDtlsClientVerifier()
0000000000134fc0 T QDtlsClientVerifier::~QDtlsClientVerifier()
0000000000134fc0 T QDtlsClientVerifier::~QDtlsClientVerifier()
000000000015ade0 t QDtlsPrivateOpenSSL::verifyPeer()
0000000000157b90 t QDtlsPrivateOpenSSL::handleTimeout(QUdpSocket*)
00000000001599a0 t QDtlsPrivateOpenSSL::reportTimeout()
0000000000157a10 t QDtlsPrivateOpenSSL::TimeoutHandler::timerEvent(QTimerEvent*)
0000000000159500 t QDtlsPrivateOpenSSL::TimeoutHandler::doubleTimeout()
0000000000159520 t QDtlsPrivateOpenSSL::TimeoutHandler::stop()
00000000001594e0 t QDtlsPrivateOpenSSL::TimeoutHandler::start(int)
000000000015c230 t QDtlsPrivateOpenSSL::TimeoutHandler::~TimeoutHandler()
000000000015c210 t QDtlsPrivateOpenSSL::TimeoutHandler::~TimeoutHandler()
000000000015c210 t QDtlsPrivateOpenSSL::TimeoutHandler::~TimeoutHandler()
000000000015acb0 t QDtlsPrivateOpenSSL::abortHandshake(QUdpSocket*)
000000000015a520 t QDtlsPrivateOpenSSL::startHandshake(QUdpSocket*, QByteArray const&)
0000000000158be0 t QDtlsPrivateOpenSSL::decryptDatagram(QUdpSocket*, QByteArray const&)
000000000015bae0 t QDtlsPrivateOpenSSL::resumeHandshake(QUdpSocket*)
000000000015bd80 t QDtlsPrivateOpenSSL::continueHandshake(QUdpSocket*, QByteArray const&)
000000000015b650 t QDtlsPrivateOpenSSL::pskClientCallback(char const*, char*, unsigned int, unsigned char*, unsigned int)
000000000015b370 t QDtlsPrivateOpenSSL::pskServerCallback(char const*, unsigned char*, unsigned int)
000000000015ad40 t QDtlsPrivateOpenSSL::sendShutdownAlert(QUdpSocket*)
000000000015b210 t QDtlsPrivateOpenSSL::storePeerCertificates()
0000000000158910 t QDtlsPrivateOpenSSL::writeDatagramEncrypted(QUdpSocket*, QByteArray const&)
[... 67 lines cut ...]
00000000001bf270 d typeinfo for QDtlsPrivateOpenSSL
00000000001bf240 d typeinfo for QDtlsClientVerifierOpenSSL
00000000001bf210 d typeinfo for QDtlsClientVerifierPrivate
00000000001c1768 V typeinfo for QDtls
00000000001bf258 d typeinfo for QDtlsPrivateOpenSSL::TimeoutHandler
000000000017db70 r typeinfo name for QDtlsPrivate
000000000017db30 r typeinfo name for QDtlsBasePrivate
0000000000185380 V typeinfo name for QDtlsClientVerifier
000000000017dbd0 r typeinfo name for QDtlsPrivateOpenSSL
000000000017db80 r typeinfo name for QDtlsClientVerifierOpenSSL
000000000017db50 r typeinfo name for QDtlsClientVerifierPrivate
0000000000185396 V typeinfo name for QDtls
000000000017dba0 r typeinfo name for QDtlsPrivateOpenSSL::TimeoutHandler
00000000001bf2b0 d vtable for QDtlsPrivate
00000000001bc908 d vtable for QDtlsBasePrivate
00000000001c1780 V vtable for QDtlsClientVerifier
00000000001bc950 d vtable for QDtlsPrivateOpenSSL
00000000001bc928 d vtable for QDtlsClientVerifierOpenSSL
00000000001bf288 d vtable for QDtlsClientVerifierPrivate
00000000001c17f0 V vtable for QDtls
00000000001bf310 d vtable for QDtlsPrivateOpenSSL::TimeoutHandler
00000000001599b0 t q_X509DtlsCallback

MyHost{sekigawa}(3)$
```

Kazzz-S, 2024-01-25:
**6. Rebuild of the commit in question for Linux with the above `Qt5.15.7`**
I could successfully build the commit on Linux Mint 20.3 using...
```
MyHost{sekigawa}(1)$ clang++ --version
clang version 17.0.6 (https://github.com/llvm/llvm-project.git 6009708b4367171ccdbf4b5905cb6a803753fe18)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/local/bin
```
[image]
[image]
[image]

Kazzz-S, 2024-01-25:
**7. Conclusions and Actions**
I have submitted [the above ticket](https://github.com/Homebrew/brew/issues/16529).
Let's wait for the response.

Kazzz-S, 2024-01-25:
**8. Using `Homebrew Qt6.6.1`**
Bo98 commented in the future request ticket above:
```
Qt6 supports multiple backends so both OpenSSL and SecureTransport are enabled there
and it supports DTLS if OpenSSL is used (the default).
```

So, I tried to build another commit in question with `Homebrew Qt6.6.1`, and found it successful.
```
(base) MacBookPro2{sekigawa} QtNetwork (1)% pwd
/Users/sekigawa/GitWork/klayout/qt6Brew.build.macos-Monterey-debug-Rhb33Phb311/gsiqt/qt6/QtNetwork

(base) MacBookPro2{sekigawa} QtNetwork (2)% ll | grep Dtls
-rw-r--r--  1 sekigawa staff 3028800  1 25 18:36 gsiDeclQDtls.o
-rw-r--r--  1 sekigawa staff 1940608  1 25 18:36 gsiDeclQDtlsClientVerifier.o
-rw-r--r--  1 sekigawa staff 1400696  1 25 18:36 gsiDeclQDtlsClientVerifier_GeneratorParameters.o
-rw-r--r--  1 sekigawa staff 1917216  1 25 18:36 gsiDeclQDtlsError.o
```

But I cannot quit KLayout GUI application normally. I need to kill the process.