KLayout/klayout #1610: KLayout Add Technology bug?

lukasc-ubc, 2024-02-04:
Hi @klayoutmatthias 

Here is a video of a strange bug, in 0.28.15. 

The steps are:

1. Download this Technology PDK: https://github.com/SiEPIC/SiEPICfab-EBeam-ZEP-PDK
2. Start KLayout 
3. Tools > Manage Technologies, Import Technology.  Then find the SiEPICfab_EBeam_ZEP.lyt file.
4. Check "General" Technology specific libraries, and notice that it is empty
5. Create new layout using the ZEP technology, and notice that there are no libraries
6. Macros > Macro Development: missing the "[Technology SiEPICfab_EBeam_ZEP]" folder
7. Restart KLayout
8. Repeat Steps 4, 5, 6, and notice that the libraries are there.

https://www.youtube.com/watch?v=osE6fwB9x00

@Hang-Bobby-Zou and I checked this on Mac OSX and on Windows 64 bit, and obtain exactly the same problem.

We found this error appearing when running scripts.  Specifically, it was a Python script that created a layout.  We found that even if the libaries are present (after Step 8), we would intermittently find that sometimes the script fails because it cannot load the Libraries.  We couldn't create a minimum reproducible example unfortunately.  Six of us had the same issue, but now it is working for me again.

Ultimately, we found that 0.28.12 didn't have this intermittent issue.  (yet the strange bug in the video also takes place in 0.28.12)...

thank you
Lukas

klayoutmatthias, 2024-02-04:
Hi Lukas,

thanks for this nicely prepared test case.

Initially I could not see any libraries at all. The reason was that I did not have "siepic_tools" installed:

```
SiEPICfab_EBeam_ZEP_Library
ERROR: /home/matthias/klayout/testdata/SiEPICfab-EBeam-ZEP-PDK-master/SiEPICfab_EBeam_ZEP/pymacros/SiEPICfab_EBeam_ZEP_Library.lym:27: ModuleNotFoundError: No module named 'SiEPIC'
  /home/matthias/klayout/testdata/SiEPICfab-EBeam-ZEP-PDK-master/SiEPICfab_EBeam_ZEP/pymacros/SiEPICfab_EBeam_ZEP_Library.lym:27 (class ModuleNotFoundError)
SiEPICfab_EBeam_ZEP_Library
ERROR: /home/matthias/klayout/testdata/SiEPICfab-EBeam-ZEP-PDK-master/SiEPICfab_EBeam_ZEP/pymacros/SiEPICfab_EBeam_ZEP_beta_Library.lym:27: ModuleNotFoundError: No module named 'SiEPIC'
  /home/matthias/klayout/testdata/SiEPICfab-EBeam-ZEP-PDK-master/SiEPICfab_EBeam_ZEP/pymacros/SiEPICfab_EBeam_ZEP_beta_Library.lym:27 (class ModuleNotFoundError)
```

After installing the package, I see this error, but I guess that is not related to this problem:

```
ERROR: /home/matthias/.klayout-issue-1610/salt/siepic_tools/tech/GSiP/pymacros/GSiP Library.lym:113: RuntimeError: Unable to open file: /home/matthias/.klayout-issue-1610/salt/siepic_tools/tech/GSiP/gds/building_blocks//grating_coupler_13deg_te_1550_oxide.gds (errno=2) in Layout.read
  /home/matthias/.klayout-issue-1610/salt/siepic_tools/tech/GSiP/pymacros/GSiP Library.lym:113
  /home/matthias/.klayout-issue-1610/salt/siepic_tools/tech/GSiP/pymacros/GSiP Library.lym:134 (class RuntimeError)
```

Immediately after importing the .lyt file I do not see the libraries, but I guess that upon importing .lyt the macros are not executed. Only after restarting, the macros are executed and the libraries become available. 

After restarting once, everything seems to be in place. 

I understand however, that in your case, this is not always true. So you mean when you restart KLayout, sometimes the libraries are not present in that technology?

Have you checked for errors in the log like the one about the missing "SiEPIC" module?

Matthias

lukasc-ubc, 2024-02-05:
Hi Matthias,

The GSiP error is ok (at least unrelated to this issue).

The real problem was as follows:

- Open KLayout, the ZEP technology is available
- Run a script that creates a new layout
```
tech="SiEPICfab_Shuksan_PDK"
mw = pya.Application().instance().main_window()
ly = mw.create_layout(tech, 1).layout()
...
```
- the script attempts to instantiate components, but fails because the Libraries have dissappeared

Unfortunately, I can't reproduce the error right now on Mac OSX (annoyingly intermittent).  There were 5 others running Windows that consistently had the error.  Then we debugged and backtracked to a potential cause where the libraries are not present (above). One Windows user also showed me a case where "Import Technology" didn't result in the library being added.

I will keep an eye on this and hopefully the error comes up again, and report back.  I'll get a Windows VM and test it there.

Ideally I'll come up with a simpler test that has the same problem.

klayoutmatthias, 2024-02-06:
Hi Lukas,

Here is my script, but as expected, it works and creates a piece of a waveguide:

```
tech="SiEPICfab_EBeam_ZEP"
mw = pya.Application().instance().main_window()
ly = mw.create_layout(tech, 1).layout()

top = ly.create_cell("TOP")
ci2 = ly.create_cell("Waveguide", "SiEPICfab_EBeam_ZEP", {})
top.insert(pya.CellInstArray(ci2, pya.Trans()))

mw.current_view().active_cellview().cell = top
mw.current_view().zoom_fit()
```

Maybe some comments are helpful from my side for further debugging.

First, the libraries are not really tied to a technology. A library does not become entirely unavailable, if the technology is not set properly. At least it should be listed in the library names:

```
pya.Library.library_names()
```

So if you see the cells cannot be instantiated, you can check first, if the library is listed under "names". If not, the library registration has failed in general. If it is listed, but cells cannot be instantiated, the technology binding may not be correct.

If technology binding turns out not to be happening, you can check:

1. if the layout uses the proper technology: `ly.technology().name`
2. if the library is tied to that technology: `pya.Library.library_by_name(library_name, tech_name)` must not return None

If both conditions are met, the technology binding should work.

Matthias

lukasc-ubc, 2024-02-09:
Hi Matthias,

Here is a case where the two conditions are met, yet the Library has disappeared from the window.

[image]

Intermittent unfortunately.

klayoutmatthias, 2024-02-10:
Perfect! I think I can reproduce the issue.

I will debug the problem. Looks like the debugger mode supresses some events that are necessary to keep the library linking alive. I will try to debug the issue.

Thanks,

Matthias

klayoutmatthias, 2024-02-10:
I found the issue.

It wasn't related to technology. This effect was actually a side effect. The real reason was that the layout was "under_construction" when the script ended.

The reason for that was something entirely different. It was related to the parent instance iterators which lock the layout while iteration happens. In some places these iterators are kept in variables which is perfectly fine. However, there was a leaking reference in the debugger code that held a reference to the execution frame and that prevented these variables to get released - holding the layout lock forever.

It is very easy to fix and it looks like that solves the issue.

Thanks reporting the problem and narrowing down the problem.

Matthias

lukasc-ubc, 2024-02-10:
Fantastic!  Our tape-out is this weekend, but it will be useful for next time!

lukasc-ubc, 2024-02-10:
I notice you have automated Actions that compile KLayout.  Is there a Mac version that gets created automatically?

klayoutmatthias, 2024-02-11:
Hi Lukas,

very good ... as of now the workaround is to run without debugger. That is also a lot faster :)

The Mac GUI versions are made by @Kazzz-S. Automatic deployment is always a pain with Qt included (limited resources, never the right version installed etc.). I have GitHub actions only for the Python modules, but Mac is included there.

Matthias

closed 2024-02-11 by lukasc-ubc