KLayout/klayout #1619: -e:1: [BUG] Segmentation fault at 0x0000000000000010

gcharalampous, 2024-02-11:
I have built sucesfuly klayout-0.28.15, and klayout-0.28.6 on Fedora-39. When I try to run the software I get the following segmentation error on my terminal. Please note that my previous version build still works. My flatpak version works too.

./klayout 
-e:1: [BUG] Segmentation fault at 0x0000000000000010
ruby 3.2.2 (2023-03-30 revision e51014f9c0) [x86_64-linux]

-- Control frame information -----------------------------------------------
c:0003 p:---- s:0010 e:000009 CFUNC  :__run_app__
c:0002 p:0003 s:0006 e:000005 EVAL   -e:1 [FINISH]
c:0001 p:0000 s:0003 E:000f20 DUMMY  [FINISH]

-- Ruby level backtrace information ----------------------------------------
-e:1:in `<main>'
-e:1:in `__run_app__'

-- Machine register context ------------------------------------------------
 RIP: 0x00007ffaf33e4601 RBP: 0x00007ffed145fc30 RSP: 0x00007ffed145fbc0
 RAX: 0x0000000000000000 RBX: 0x0000000000000044 RCX: 0x00007ffae72db5b0
 RDX: 0x00007ffae72da6d0 RDI: 0x0000000000000001 RSI: 0x00007ffae72d8010
  R8: 0x0000000000000000  R9: 0x0000000000000007 R10: 0x00007ffaf320b4f8
 R11: 0x00007ffaf3407090 R12: 0x0000000000000000 R13: 0x00007ffaf37de2e0
 R14: 0x0000000000000001 R15: 0x0000000006e78b1b EFL: 0x0000000000010202

-- C level backtrace information -------------------------------------------
/lib64/libruby.so.3.2(0x7ffaf2edeb99) [0x7ffaf2edeb99]
/lib64/libruby.so.3.2(0x7ffaf2ee2ed9) [0x7ffaf2ee2ed9]
/lib64/libruby.so.3.2(0x7ffaf2d3225c) [0x7ffaf2d3225c]
/lib64/libruby.so.3.2(0x7ffaf2e56b16) [0x7ffaf2e56b16]
/lib64/libc.so.6(__restore_rt+0x0) [0x7ffae4e5c9a0]
/lib64/libpython3.12.so.1.0(0x7ffaf33e4601) [0x7ffaf33e4601]
/lib64/libpython3.12.so.1.0(PyUnicode_New+0x69) [0x7ffaf33eb729]
/lib64/libpython3.12.so.1.0(0x7ffaf33eab2c) [0x7ffaf33eab2c]
/opt/klayout-0.28.6/libklayout_pya.so.0(_ZN3pya13c2python_funcIRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEclES8_+0x15) [0x7ffaef575945]
/opt/klayout-0.28.6/libklayout_pya.so.0(_ZN3pya17PythonInterpreterC2Eb+0x3c2) [0x7ffaef571cf2]
/opt/klayout-0.28.6/libklayout_lay.so.0(_ZN3lay15ApplicationBase8init_appEv+0x191f) [0x7ffaef72963f]
./klayout(0x40472b) [0x40472b]
/opt/klayout-0.28.6/libklayout_rba.so.0(0x7ffaefe264d0) [0x7ffaefe264d0]
/lib64/libruby.so.3.2(0x7ffaf2ebeefd) [0x7ffaf2ebeefd]
/lib64/libruby.so.3.2(0x7ffaf2ec2caf) [0x7ffaf2ec2caf]
/lib64/libruby.so.3.2(0x7ffaf2ec4a39) [0x7ffaf2ec4a39]
/lib64/libruby.so.3.2(rb_vm_exec+0xc2) [0x7ffaf2eda652]
/lib64/libruby.so.3.2(0x7ffaf2d3b10e) [0x7ffaf2d3b10e]
/lib64/libruby.so.3.2(ruby_run_node+0x93) [0x7ffaf2d3f053]
/opt/klayout-0.28.6/libklayout_rba.so.0(_ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E+0xc0) [0x7ffaefe27a20]
./klayout(0x404401) [0x404401]
./klayout(0x4039b2) [0x4039b2]
/lib64/libc.so.6(__libc_start_call_main+0x7a) [0x7ffae4e4614a]
/lib64/libc.so.6(__libc_start_main+0x8b) [0x7ffae4e4620b]
./klayout(0x403a45) [0x403a45]

-- Other runtime information -----------------------------------------------

* Loaded script: -e

* Loaded features:

    0 enumerator.so
    1 thread.rb
    2 fiber.so
    3 rational.so
    4 complex.so
    5 ruby2_keywords.rb
    6 /usr/lib64/ruby/enc/encdb.so
    7 /usr/lib64/ruby/enc/trans/transdb.so
    8 /usr/lib64/ruby/rbconfig.rb
    9 /usr/share/rubygems/rubygems/compatibility.rb
   10 /usr/share/rubygems/rubygems/defaults.rb
   11 /usr/share/rubygems/rubygems/deprecate.rb
   12 /usr/share/rubygems/rubygems/errors.rb
   13 /usr/share/rubygems/rubygems/unknown_command_spell_checker.rb
   14 /usr/share/rubygems/rubygems/exceptions.rb
   15 /usr/share/rubygems/rubygems/basic_specification.rb
   16 /usr/share/rubygems/rubygems/stub_specification.rb
   17 /usr/share/rubygems/rubygems/platform.rb
   18 /usr/share/rubygems/rubygems/util/list.rb
   19 /usr/share/rubygems/rubygems/version.rb
   20 /usr/share/rubygems/rubygems/requirement.rb
   21 /usr/share/rubygems/rubygems/specification.rb
   22 /usr/share/rubygems/rubygems/defaults/operating_system.rb
   23 /usr/share/rubygems/rubygems/util.rb
   24 /usr/share/rubygems/rubygems/dependency.rb
   25 /usr/share/rubygems/rubygems/core_ext/kernel_gem.rb
   26 /usr/lib64/ruby/monitor.so
   27 /usr/share/ruby/monitor.rb
   28 /usr/share/rubygems/rubygems.rb
   29 /usr/share/rubygems/rubygems/path_support.rb
   30 /usr/share/ruby/error_highlight/version.rb
   31 /usr/share/ruby/error_highlight/base.rb
   32 /usr/share/ruby/error_highlight/formatter.rb
   33 /usr/share/ruby/error_highlight/core_ext.rb
   34 /usr/share/ruby/error_highlight.rb
   35 /usr/share/ruby/did_you_mean/version.rb
   36 /usr/share/ruby/did_you_mean/core_ext/name_error.rb
   37 /usr/share/ruby/did_you_mean/levenshtein.rb
   38 /usr/share/ruby/did_you_mean/jaro_winkler.rb
   39 /usr/share/ruby/did_you_mean/spell_checker.rb
   40 /usr/share/ruby/did_you_mean/spell_checkers/name_error_checkers/class_name_checker.rb
   41 /usr/share/ruby/did_you_mean/spell_checkers/name_error_checkers/variable_name_checker.rb
   42 /usr/share/ruby/did_you_mean/spell_checkers/name_error_checkers.rb
   43 /usr/share/ruby/did_you_mean/spell_checkers/method_name_checker.rb
   44 /usr/share/ruby/did_you_mean/spell_checkers/key_error_checker.rb
   45 /usr/share/ruby/did_you_mean/spell_checkers/null_checker.rb
   46 /usr/share/ruby/did_you_mean/tree_spell_checker.rb
   47 /usr/share/ruby/did_you_mean/spell_checkers/require_path_checker.rb
   48 /usr/share/ruby/did_you_mean/spell_checkers/pattern_key_name_checker.rb
   49 /usr/share/ruby/did_you_mean/formatter.rb
   50 /usr/share/ruby/did_you_mean.rb
   51 /usr/share/ruby/syntax_suggest/core_ext.rb

* Process memory map:
00400000-00403000 r--p 00000000 00:20 2003148                            /opt/klayout-0.28.6/klayout
00403000-00405000 r-xp 00003000 00:20 2003148                            /opt/klayout-0.28.6/klayout
00405000-00406000 r--p 00005000 00:20 2003148                            /opt/klayout-0.28.6/klayout
00406000-00407000 r--p 00005000 00:20 2003148                            /opt/klayout-0.28.6/klayout
00407000-00408000 rw-p 00006000 00:20 2003148                            /opt/klayout-0.28.6/klayout
0161a000-07ae3000 rw-p 00000000 00:00 0                                  [heap]
7ffa70000000-7ffa70021000 rw-p 00000000 00:00 0 
7ffa70021000-7ffa74000000 ---p 00000000 00:00 0 

The list here is very big I can't post everything.

Aborted (core dumped)

klayoutmatthias, 2024-02-11:
You're saying you built it from sources yourself and simply running the software makes it crash?

There are numerous potential causes from incorrect linking, wrong library search paths, wrong ABI and such. Building software is an art and there are many ways to spoil it.

For example: your dump indicates that the application binary comes from `./klayout` while the libraries are pulled from `/opt/klayout-0.28.6/`. So I think that in your installation the search path is messed up. If `./klayout` is supposed to be 0.28.15, then it is wrong to pull libraries from 0.28.6. 

Matthias

gcharalampous, 2024-02-11:
Thank you for your response. Yes, I have built the software myself on Fedora multiple times, and this is the first time I am facing a challenge.

I have two Fedora workstations. I built KLayout-0.28.6 on my first Fedora workstation when it was initially available, and it is still working and running successfully. Recently, I attempted to build KLayout-0.28.15 on the same machine. However, when I run the software, it crashes with a segmentation error.

On my second Fedora workstation, I also attempted to build KLayout-0.28.15, and similarly, when I run it, it crashes with a segmentation error. Recognizing that KLayout-0.28.6 works on my first workstation, I tried to build this version on my second machine. Surprisingly, it also crashed with the same segmentation error.

I encounter the same error on two different machines while building two different versions. Both machines are running Fedora-39 (Wayland), and my system has the latest updates.

I did not mess up the two versions since I am not pulling libraries from 0.28.6 to run 0.28.15. I have built both versions on separated folders under `/opt/klayout-0.28.6` and `/opt/klayout-0.28.15` directories. Then I `cd` to `/opt/klayout-0.28.6` and type in my terminal `./klayout` .

The steps I follow to build the software;
1. I have installed all the packages required for the build from the klayout website. 
2. I am creating an empty directory `/opt/klayout-version`, different directory for each version I build.
3. I download the version I want from the website and untar in  my `~/Downloads` folder
4. I run the `build.sh` script with the following command `bash build.sh -prefix /opt/klayout-version/`
5. If the build is sucesful the software should run when I cd to `/opt/klayout-version/` and `./klayout`.

If anyone is encountering the same issue I will be interested to read their experience.

Thank you, 
Georgios

klayoutmatthias, 2024-02-11:
Hi @gcharalampous,

A useful approach is to run the build inside a Docker container based on a standard image. This is what I am doing regularly on a number of different Linux distributions. I did not include Fedora as of now. A Docker container is a way to create a reproducible environment that allows me debugging the issue.

I have conducted that experiment with a fresh Docker image from fedora:39 and I think I can reproduce the issue.

At first glance the problem comes from Python 3.12.1. Unlike other versions, this version does not allow using `PyUnicode_DecodeUTF8` before the application is initialized through `Py_Initialize`. Bad if you have to convert UTF-8 paths to wide UTF ones ...

I created a patch using KLayout's conversion routines here: https://github.com/KLayout/klayout/commit/e96917345af28d498571eba79806a61afbea1e22

It seems to fix the particular issue on Python 3.12.1, but awaits verification now.

Matthias

gcharalampous, 2024-02-13:
Thank you, @klayoutmatthias. I appreciate your assistance. I successfully built and ran KLayout after your latest commit. Although I'm not well-versed in Docker, I'm willing to give it a try.

I observed that you have included some notes under the `ci-scripts` directory. The reason I haven't explored Docker until now is due to a previous negative experience with the flatpak version. The flatpak version, which also operates KLayout as a container, did not effectively communicate with other software crucial for building my .GDS files. For instance, the klive package failed to interact with my Python IDE. This issue is not present when I build KLayout locally.

I can give Docker a try since you recommended it.

closed 2024-02-13 by gcharalampous

klayoutmatthias, 2024-02-13:
Hi @gcharalampous,

I think you should give Docker a try. It's not the same as the sandbox environment flatpak runs inside. It's an entire system with it's own network namespace and file system. Inside a Docker container you have root privileges and can install any packages without harm to the host system. It's not good for running a UI application IMHO, but it is good for building software in a known configuration.

For deployment you can simply copy the artefacts from Docker to your host system. I'd also like to check AppImage which I think is a more lightweight alternative to flatpak or snap and sandboxing is optional there (I dislike that too).

Matthias

pwuertz, 2024-05-17:
Ubuntu 24.04 now installs `klayout-0.28.15` by default, which crashes on start-up with the reported error.

If I read this correctly, @klayoutmatthias, identified this to be a `PyUnicode_DecodeUTF8` before `Py_Initialize` error, which Python 3.12 doesn't tolerate anymore. The fix for this is 1eefbf35add973bbb9120795b908342a250644eb, which is included in versions `>=0.28.16`, correct?

klayoutmatthias, 2024-05-17:
Thanks for bringing this up. I am still on Ubuntu 22 myself, but I will try myself asap.

Matthias

klayoutmatthias, 2024-05-17:
I just tried on Ubuntu 24 inside a Docker container.

I can confirm that installing klayout (0.28.15) from the repo renders a dysfunctional installation and I can reproduce the crash. I also can reproduce it with a self-built version 0.28.15.

In addition, I confirmed that 0.28.16 fixes the problem. I'd recommend 0.28.17 as the latest 0.28 branch version. Or to switch to 0.29.1 which is also working.

Matthias

pwuertz, 2024-05-18:
Great, seeing good results with 0.29.1 as well.

Bug report for Ubuntu is at https://bugs.launchpad.net/ubuntu/+source/klayout/+bug/2066026 in case someone likes to add a +affectsmetoo.