KLayout/klayout #1625: Technology-specific Default Grids

Kazzz-S, 2024-02-14:
Original Post => https://www.klayout.de/forum/discussion/2459/

This enhancement feature does not seem to be included in `0.28.16`.
Please include it in a future release :-)

I manually merged the `tech-specific-grids` branch into `0.28.16` on Linux and found it OK.
[image]

Best regards,
Kazzz-S

klayoutmatthias, 2024-02-14:
Oh, I'm terribly sorry ... I simply forgot that :(

I will make a new release quickly.

Matthias

Kazzz-S, 2024-02-14:
Thank you very much. But I'm sorry I seemed to have rushed you.

Kazzz-S

closed 2024-02-15 by klayoutmatthias

klayoutmatthias, 2024-02-17:
No worries, you did not rush me! :)

I just lost track of the many open topics. I should have created a ticket for tracking.

But I hope 0.28.17 is now fixing this. Last thing I had to add was scripting support for the new technology attribute, so the feature is complete now.

Best regards,

Matthias

Kazzz-S, 2024-02-17:
Thanks a lot for releasing 0.28.17.
I'm building that version on macOS, as usual. **The 1st set on Monterey is over**.
[image]

Best regards,
Kazzz-S