KLayout/klayout #1631: [mac only?] Weird Menu Strings

Kazzz-S, 2024-02-24:
This problem was reported by @stefanottili at least twice:

1. https://github.com/KLayout/klayout/issues/1452#issuecomment-1668409686
2. https://github.com/KLayout/klayout/issues/1602#issuecomment-1937753480

I could reproduce the problem on Monterey (Intel Mac) as follows.

----

**Build and Deploy**
* Using Qt 6.6.2 from Homebrew,
```
% ./build4mac.py -q qt6brew -r hb33 -p hb311  (build)

% ./build4mac.py -q qt6brew -r hb33 -p hb311  -Y (deploy)
```

----

**Test-1: direct start of the executable**
1. Copy the attached [macQAT.py](https://github.com/KLayout/klayout/files/14391534/macQAT.zip)  script to the `qt6Brew.bin.macos-Monterey-release-Rhb33Phb311/` directory.
2. Move to the directory.
3. Then, invoke `./macQAT.py -k`, where
   * the `DYLD_LIBRARY_PATH` environment variable is set appropriately.
   * the executable `./klayout.app/Contents/MacOS/klayout` is invoked.

The outputs are as follows.
[image]
[image]
[image]

* All attempts (more than 10 times) started with Case-1. However, Case-2 occurred occasionally during a session (conditions not determinable).

----

**Test-2: open the application bundle**
1. Move to the `LW-qt6Brew.pkg.macos-Monterey-release-Rhb33Phb311` directory.
2. Then, invoke `open ./klayout.app/`
[image]

* Normal in all attempts.

Kazzz-S

klayoutmatthias, 2024-02-25:
My guess is that for some reason MacOS picks the "Ruby program name" for the menus.

KLayout calls a function named "rb_set_progname" from the embedded Ruby interpreter. It does so that Ruby reports error messages in the proper context.

Maybe on MacOS, Ruby somehow passes this name to the system which makes it think that the program is called "<immediate>" once a macro has been executed from the console.

Matthias

klayoutmatthias, 2024-02-26:
After checking the Ruby source code I believe that the patch provided in the PR for this issue may fix the issue.

Apparently Ruby decided not only to change the internal program name, but also the process name when setting the "PROGRAM_NAME" global variable. There is an alternative function which appears to do the same thing without this side effect.

The important hint was the "<immediate>" string.

Maybe you can check this and give feedback whether the problem is fixed?

Thanks,

Matthias

klayoutmatthias, 2024-02-26:
OMG ... hard to believe it was that simple.

I swear I had no clue where to look at, until I say the `<immediate>` thing. That was a kind of eye opener.

Matthias

Kazzz-S, 2024-02-26:
Hi @klayoutmatthias,

Thank you for your efforts to resolve this issue.
I've re-run the same Test-1 and Test-2 above with the PR branch and confirmed that the issue has been resolved :-)

Best regards,
Kazzz-S

closed 2024-03-09 by klayoutmatthias