KLayout/klayout #1633: Klayout cells window is not responding

wongmatthew73, 2024-02-24:
Regarding Klayout "Cells" window, it doesn't respond when I click on it. In the screenshot, when I try and click on the spm arrow to expand the list, it doesn't respond. If I double click anywhere in the "Cells" window, the window will move behind the main program window. I then hover over the area where the "Cells" window disappeared for it to reappear/move to the front. It's as though the window is a read only graphic. Everything else in Klayout seems to work properly except this. Is the Cells window locked?

Klayout 28.2-1
Ubuntu 22.04

I've tried opening multiple different projects and this scenario happens on all of them.

[image]

sebastian-goeldi, 2024-02-25:
Can you try updating klayout? 0.28.2 is from December 2022, so more than a year old. There's a good chance that might already fix it for you. 

If it doesn't, it might be due to having a really busy gds/oas. It might help to increase the number of rendering threads or wait until everything's loaded (for me that can happen on rather big / polygon rich designs, esp with fill and I have to wait a few seconds even after the UI becomes responsive again. Though disclaimer I mostly work with photonics not electronics)

klayoutmatthias, 2024-03-21:
@wongmatthew73  Any updates on this? Did you try with the latest version?

I am using Ubuntu 22.04 myself and can't see that issue. 

You can try renaming your `~/.klayout` folder to something else. This will clear the configuration that KLayout stores to persist window dimensions and dock status. Maybe this helps.

Matthias

Madhav-Malhotra, 2024-07-16:
> @wongmatthew73 Any updates on this? Did you try with the latest version?
>
> I am using Ubuntu 22.04 myself and can't see that issue.
>
> You can try renaming your `~/.klayout` folder to something else. [...]

I'm also having the same issue on an Ubuntu 22.04.04 WSL2 instance running klayout 0.28.2. I'll try the update or clearing the configuration file.