KLayout/klayout #1651: create Polygons from lists

joamatab, 2024-03-10:
How can we speed polygon creation from list of points?

When dealing with polygons with multiple points it would be faster not having to convert to kdb.Point or kdb.DPoint

How can we make this work?

```python
import kfactory as kf 

poly1 = kf.kdb.DPolygon(
    [
        (-8, -6),
        (6, 8),
        (7, 17),
        (9, 5),
    ]
)
```

The speed up would be quite significant for large polygons like this
```python
        if isinstance(points, tuple | list | np.ndarray):
            polygon = kdb.DPolygon([kdb.DPoint(x, y) for (x, y) in points]) # works but slow

            polygon = kdb.DPolygon(np.array(points, dtype=float).tolist()) # would be faster but fails
```

related to https://github.com/KLayout/klayout/issues/1530

@amccaugh 
@sebastian-goeldi

sebastian-goeldi, 2024-03-10:
This works, but it seems the .pyi didn't pick up the new function overload.

```pycon
>>> import klayout.db as kdb
>>> kdb.Polygon([[0,0],[10,0],[10.5,10]])
(0,0;10,10;10,0)
>>> kdb.DPolygon([[0,0],[10,0],[10.,10]])
(0,0;10,10;10,0)
```

sebastian-goeldi, 2024-03-10:
The issue seems to be if the list has exactly 4 points, then the creation fails

```pycon
oly1=kf.kdb.DPolygon([(0,0),[-10,0],[10.,10],[10,0]])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Ambiguous overload variants - multiple method declarations match arguments in DPolygon.__init__
```

@klayoutmatthias could it be that klayout tries to cast to a Box first here?

klayoutmatthias, 2024-03-10:
Grml ... I knew too much smartness would finally create some trouble ... :(

Basically, the whole scheme why this is works is the inference scheme KLayout uses: if an argument expects and object, but is passed an array, KLayout tries to call the object constructor from the arguments of the array. This also works for array elements. 

This is way `(0,0)` can create a `DPoint` in the `DPolygon([DPoint, DPoint, ... ])` constructor. In the same fashion a four-element array can be matched to the arguments of `DBox` in the DPolygon(DBox)` constructor. This is what happens here.

I can fix this by giving precedence of array matching array (I think).

The best workaround is to use "assign_hull" which does not have a `Box` overload:

```
poly = kf.kdb.DPolygon()
poly.assign_hull([(0,0),[-10,0],[10.,10],[10,0]])
```

Matthias

joamatab, 2024-03-10:
awesome! that worked, thank you Matthias!

feel free to close this

klayoutmatthias, 2024-03-10:
Very good, thanks for the feedback.

I'll keep that ticket open to remind me of this stupid ambiguity. I shall be less permissive on the inference paths.

Matthias

closed 2024-03-29 by klayoutmatthias