KLayout/klayout #1665: klayout def import error

OuDret, 2024-03-27:
Hello, 

I was trying to convert large DEF design using klayout & [this klayout python interpreter script](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/def2stream.py) which created gds with no error raised but when i try to open it seems to be corrupted.

When I open DEF in openroad all seems to be alright.

[image]

Do you have any suggestion of where the problem is from this error message?

I also tried to import it manually lef/def from the UI but it failed.

I have converted many designs from def to gds using klayout python interpreter (althought none of them was that big) but never faced this error.

I will appreciate any insight.

stefanottili, 2024-03-27:
> I also tried to import it manually lef/def from the UI but it failed.

what failed ?

I would give the strm2oas/gds buddy tool a try using a map file.
First without the gds files for the lef macros, then with.

klayoutmatthias, 2024-03-27:
Is it possible to share the GDS you created?

It is hard to imagine what might cause this issue. Things like empty polygons are other border case features come to mind.

If you cannot share the entire file, a hex dump of +/-1000 bytes before and after the indicated location has helpful.

On Linux you can create such a dump with

```
od -tx1 -Ad <file.gds>
```

and then pick +/-100 lines before the indicated address like in this made-up example:

```
...
1149072 95 1d 23 1d 5b 3e 21 8f 4e 39 46 1e 25 13 76 fb
1149088 a2 e7 6b 22 47 26 9f 20 df 0e ff 07 e9 af 70 3b
1149104 66 45 05 dd 37 f2 18 d9 31 d4 65 6e cb 2d d1 b7
1149120 57 3e 4d 9a 97 ae 14 af 90 d3 b5 0f 3b 3b 86 46
1149136 de ab 14 64 f7 aa e8 f3 ab a2 07 a7 3f 4d 5e 1c
1149152 18 b5 9f 91 83 05 91 9e f2 2d 87 cb a3 2d 03 57
1149168 b2 f3 a4 37 7d 8b 1c 7c 32 3d d2 4e 27 ec c9 8c
1149184 7c 3a cb 95 4f 92 c7 73 a3 1d 43 22 ea 96 a3 a9   <!----- 1149186
1149200 5b e4 ad 2d f5 2b c5 94 96 60 f4 83 01 d1 7d c1
1149216 4f 48 77 28 fa b6 03 4e c2 d9 48 f4 dc 5c d7 6c
1149232 f8 8e 46 ce 2c 98 f0 7c 43 e4 5c 76 94 1d 22 63
1149248 27 ec 21 2b f7 ae 99 a0 df 73 2e df 14 f9 2e e8
1149264 9a 9f fa 9a c7 45 9b c7 fe c9 b1 29 76 00 2c ef
1149280 fd 24 7a 6e e5 63 a0 4c 76 85 26 a8 9a 17 2b 56
...
```

Matthias

stefanottili, 2024-03-27:
Use https://github.com/klayoutmatthias/dump_oas_gds2 to debug gds issues.
Can you read all the individual leaf cell gds into klayout/strm2gds ?

OuDret, 2024-03-28:
I have been trying some of your suggestions.

First of all here is the hexdump of the "corrupted" address when i generate gds through klayout python interpreter:
```
1149072 00 06 0e 02 00 01 00 06 21 02 00 02 00 08 0f 03
1149088 00 00 9c 40 00 14 10 03 07 2b 57 a8 03 63 4f a4
1149104 07 2b 57 a8 03 78 10 24 00 04 11 00 00 04 09 00
1149120 00 06 0d 02 00 02 00 06 0e 02 00 01 00 06 21 02
1149136 00 02 00 08 0f 03 00 00 9c 40 00 14 10 03 07 74
1149152 95 a8 03 63 4f a4 07 74 95 a8 03 76 d7 a4 00 04
1149168 11 00 00 04 09 00 00 06 0d 02 00 02 00 06 0e 02   <!----- 1149186
1149184 00 01 00 06 21 02 00 02 00 08 0f 03 00 00 9c 40
1149200 00 14 10 03 07 ac bc a8 03 63 4f a4 07 ac bc a8
1149216 03 78 10 24 00 04 11 00 00 04 09 00 00 06 0d 02
1149232 00 02 00 06 0e 02 00 01 00 06 21 02 00 02 00 08
1149248 0f 03 00 00 9c 40 00 14 10 03 07 86 e5 28 03 63
1149264 4f a4 07 86 e5 28 03 78 10 24 00 04 11 00 00 04
1149280 09 00 00 06 0d 02 00 02 00 06 0e 02 00 01 00 06
```
I have also tried to use strm2gds tool without cells gds 
```
 strm2gds --lefdef-map SK/SK.map \
                  --debug-level 20 \
                  --lefdef-macro-resolution-mode 2 \
                  --lefdef-no-implicit-lef \
                  --lefdef-lefs=SK/SK_stdcell.lef,SK/SK_tech.lef \
                  --lefdef-joined-paths \
                  designs/serv/results/serv.def \
                  test.gds
```
```
      Reading LEF/DEF map file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK.map
      Warning: Reading layer map file SK/SK.map, line 10: purpose LABEL ignored for layer metal1
      Warning: Reading layer map file SK/SK.map, line 12: purpose LABEL ignored for layer metal3
      Warning: Reading layer map file SK/SK.map, line 14: purpose LABEL ignored for layer metal2
      Reading SK/SK_stdcell.lef
      Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK_stdcell.lef
      Reading SK/SK_tech.lef
      Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK_tech.lef
      Reading /home/joan/Desktop/Netlist2SmartKemGDS/designs/serv/results/serv.def
      Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/designs/serv/results/serv.def
      Warning: No mapping for purpose 'OUTLINE' - layer is ignored
      Reading serv.def ..
      .. 6k lines
      Warning: No mapping for layer 'metal1', purpose 'LABEL' - layer is ignored
      Warning: No mapping for layer 'metal3', purpose 'LABEL' - layer is ignored
      Warning: No mapping for layer 'metal2', purpose 'LABEL' - layer is ignored
```

It seems all routing it's fine, lef cells are not shown though.
[image]
[image]

The strangest is that running strm2gds for another design (submodule of this one) i get a gds with all cell in std_cells but not the module with my input def design.
[image]
```
strm2gds --lefdef-map SK/SK.map \
--debug-level 20 \
--lefdef-macro-resolution-mode 2 \
--lefdef-no-implicit-lef \
--lefdef-lefs=SK/SK_stdcell.lef,SK/SK_tech.lef \
--lefdef-joined-paths \
designs/serv/results/serv.def \
test.gds
Reading LEF/DEF map file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK.map
Warning: Reading layer map file SK/SK.map, line 10: purpose LABEL ignored for layer metal1
Warning: Reading layer map file SK/SK.map, line 12: purpose LABEL ignored for layer metal3
Warning: Reading layer map file SK/SK.map, line 14: purpose LABEL ignored for layer metal2
Reading SK/SK_stdcell.lef
Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK_stdcell.lef
Reading SK/SK_tech.lef
Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/SK/SK_tech.lef
Reading /home/joan/Desktop/Netlist2SmartKemGDS/designs/serv/results/serv.def
Reading LEF/DEF file /home/joan/Desktop/Netlist2SmartKemGDS/designs/serv/results/serv.def
Warning: No mapping for purpose 'OUTLINE' - layer is ignored
Reading serv.def ..
.. 6k lines
Warning: No mapping for layer 'metal1', purpose 'LABEL' - layer is ignored
Warning: No mapping for layer 'metal3', purpose 'LABEL' - layer is ignored
Warning: No mapping for layer 'metal2', purpose 'LABEL' - layer is ignored
joan@joan-646BED:~/Desktop/Netlist2SmartKemGDS$ make strmtest
strm2gds --lefdef-map SK/SK.map \
--debug-level 20 \
--lefdef-macro-resolution-mode 2 \
--lefdef-no-implicit-lef \
--lefdef-lefs=SK/SK_stdcell.lef,SK/SK_tech.lef \
--lefdef-joined-paths \
designs/serv_alu/results/serv_alu.def \
test.gds
```

OuDret, 2024-03-28:
> > I also tried to import it manually lef/def from the UI but it failed.
> 
> what failed ?

The error shown when I try to import  from klayout gui:
[image]
[image]
I might be doing something wrong when importing manually...
```
# serv.def
87 GCELLGRID X 0 DO 216 STEP 600000 ;
88 GCELLGRID Y 0 DO 216 STEP 600000 ;
89 COMPONENTS 8099 ;
90    - 0_2 EMPTY + FIXED ( 0 3540000 ) N ;
91    - 0_22_2 EMPTY + FIXED ( 660000 3540000 ) N ;
92  - 0_22_24_2 EMPTY + FIXED ( 1320000 3540000 ) N ;
```
I have EMPTY cell boths in gds library and LEF file, where it's the first cell to be defined:
[image]

stefanottili, 2024-03-28:
The gds read error is in "serv", so I would assume that some def net geometry is causing an issue.
Since you have a reliable way to reproduce it using the python script, you'll have to do some binary search.
The beauty of def as a text format is that you only need a text editor.
Comment out or delete half the NETS or SPECIALNETS, see whether the error persists, rinse and repeat.
You should be able to find which net is causing you issue relatively quickly.

klayoutmatthias, 2024-03-28:
Hi @OuDret,

I think the dump is not from the GDS, but DEF I think. Please dump the GDS file and some more lines.

But I have not clue what is wrong with the DEFs. I need the files. You can send it to me privately (contact 'at' klayout.de) in case you don't want to share it here.

Matthias

OuDret, 2024-04-03:
I will try binary search, if I don't find the error will contact you privately. Thank you all for the help.

closed 2024-04-03 by OuDret