KLayout/klayout #1670: Older klayout version disappeared from pip

urish, 2024-04-01:
For instance, version 0.28.12.

The following command worked this morning:

```
pip install klayout==0.28.12
```

but now it fails with:

```
ERROR: Could not find a version that satisfies the requirement klayout==0.28.12 (from versions: 0.26.12, 0.27.13, 0.28.17.post1, 0.29.0)
ERROR: No matching distribution found for klayout==0.28.12
```

Also, this is a good opportunity to say thanks - I'm using KLayout nearly on a daily basis, and it's a wonderful tool.

klayoutmatthias, 2024-04-01:
Yes, that is because PyPI limits the project space to 10GB. I had to delete versions as I could not install the latest 0.29.0.

Matthias

urish, 2024-04-01:
Thanks for explaining! So the policy going forward is to keep the latest patch version of each major release?

klayoutmatthias, 2024-04-01:
My recommendation is 0.28.17 in your case, or you try 0.29.0 (feedback appreciated).

I am sorry if that spoils your setup. 0.28.17 should be quite the same than 0.28.12. I could not tell which ones to 
keep, so I just left the latest one.

Matthias

urish, 2024-04-01:
Tried 0.29.0 - seems like it work well so far.

I also maintain a [VM with tools for analog designers on sky130](https://github.com/TinyTapeout/analog-virtualbox-vm-sky130a/), and and have just updated klayout to 0.29.0 there.

closed 2024-04-01 by urish

klayoutmatthias, 2024-04-01:
Very good. Thanks! :)

I promise I will not do this more often than absolutely required. Maybe there is a way to ask PyPI for more space. Unfortunately every release covers many Python versions and platforms. That takes a lot of space now.

Matthias

urish, 2024-04-01:
Looks like this repo is how you ask for more space: https://github.com/pypi/support?tab=readme-ov-file#issues-which-should-be-opened-here. Looks like it takes a couple of weeks until you get a response, e.g. https://github.com/pypi/support/issues/3445

klayoutmatthias, 2024-04-01:
That is good. Thanks for the pointer. I guess I should first check my options to make a "best effort" attempt to reduce the file sizes.

Best regards,

Matthias