KLayout/klayout #1676: Mac DMGs for 0.29.0

Kazzz-S, 2024-04-06:
Dear Sir,

Thank you for the new major release.
As usual, different DMG packages of **KLayout 0.29.0 for Mac** are available as described below ([0.29.0 Dropbox Link](https://www.dropbox.com/scl/fo/82kol9lcu24rhdvhvh58l/h?rlkey=d5kmyow853ol7r8eriocaqme0&dl=0)).
Please note that in this release:
1. `LW*Phb311.dmg` uses `Qt6.6 from Homebrew` to address  [the compilation issue](https://github.com/KLayout/klayout/issues/1599).
2. `HW*Phb311.dmg` embeds `Qt6.4 from MacPorts` to address  [the compilation issue](https://github.com/KLayout/klayout/issues/1599) and the deployment problem in `Qt6.6 from Homebrew`.
3. Both are **entirely experimental**, as the migration to Qt6 is ongoing and incomplete. In my observations, the Qt6 versions (both macOS and Linux) cannot run some Qt-based Python macros that run fine in Qt5.
Moreover, in Linux, the 2.5D view does not start.

> [!NOTE]
> Please keep this ticket open till the next maintenance release since `0.29.0` is a new major release.

## DMG file naming convention ##
[image]

where
(a) Type of package:  ST-:standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt[5|6] type:  qt[5|6]MP: MacPorts; qt[5|6]Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp33: Ruby3.3 from MacPorts; hb33: Ruby3.3 from Homebrew; ana3: Ruby3.2 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp311: Python 3.11 from MacPorts; hb311: Python 3.11 from Homebrew; ana3: Python3.11 from Anaconda3

## For the primary platform, macOS Monterey (12.7.4) ##
Five macOS DMGs for **Monterey** are available.

**Python is (3.11, 3.11, 3.11, 3.9)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
**Ruby is (3.3, 3.3, 3.2, 2.6)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
10|Monterey (12.7) |ST-klayout-0.29.0-macOS-Monterey-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(10)|
11|Monterey (12.7) with MacPorts dev. env.|LW-klayout-0.29.0-macOS-Monterey-1-qt5MP-Rmp33Pmp311.dmg|MacPorts|MacPorts|MacPorts|(11)|
12|Monterey (12.7) with Homebrew dev. env.|LW-klayout-0.29.0-macOS-Monterey-1-qt6Brew-Rhb33Phb311.dmg|Homebrew|Homebrew|Homebrew|(12)|
13|Monterey (12.7) with Anaconda3 dev. env.|LW-klayout-0.29.0-macOS-Monterey-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(13)|
14|Monterey (12.7)|HW-klayout-0.29.0-macOS-Monterey-1-qt6MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(14)|

(10) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(11) shares MacPorts development environment; for those who have the matching dev. env.
(12) shares Homebrew development environment; for those who have the matching dev. env.
(13) shares Anaconda3 development environment; for those who have the matching dev. env.
(14) shares OS-bundled Ruby2.6; Qt6 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For macOS Ventura (13.6.6) ##
Five macOS DMGs for **Ventura** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.5.1 on Monterey (12.7.4).

**Python is (3.11, 3.11, 3.11, 3.9)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
**Ruby is (3.3, 3.3, 3.2, 2.6)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
20|Ventura (13.6) |ST-klayout-0.29.0-macOS-Ventura-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(20)|
21|Ventura (13.6) with MacPorts dev. env.|LW-klayout-0.29.0-macOS-Ventura-1-qt5MP-Rmp33Pmp311.dmg|MacPorts|MacPorts|MacPorts|(21)|
22|Ventura (13.6) with Homebrew dev. env.|LW-klayout-0.29.0-macOS-Ventura-1-qt6Brew-Rhb33Phb311.dmg|Homebrew|Homebrew|Homebrew|(22)|
23|Ventura (13.6) with Anaconda3 dev. env.|LW-klayout-0.29.0-macOS-Ventura-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(23)|
24|Ventura (13.6)|HW-klayout-0.29.0-macOS-Ventura-1-qt6MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(23)|

(20) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(21) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(22) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(23) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(24) **experimental**; shares OS-bundled Ruby2.6; Qt6 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For the latest macOS Sonoma (14.4.1) ##
Five macOS DMGs for **Sonoma** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.5.1 on Monterey (12.7.4).

**Python is (3.11, 3.11, 3.11, 3.9)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
**Ruby is (3.3, 3.3, 3.2, 2.6)=(MacPorts, Homebrew, Anaconda3, OS-bundled)**
|Srl.No.|Target OS|DMG file name|Qt5 or Qt6|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
30|Sonoma (14.4) |ST-klayout-0.29.0-macOS-Sonoma-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(30)|
31|Sonoma (14.4) with MacPorts dev. env.|LW-klayout-0.29.0-macOS-Sonoma-1-qt5MP-Rmp33Pmp311.dmg|MacPorts|MacPorts|MacPorts|(31)|
32|Sonoma (14.4) with Homebrew dev. env.|LW-klayout-0.29.0-macOS-Sonoma-1-qt6Brew-Rhb33Phb311.dmg|Homebrew|Homebrew|Homebrew|(32)|
33|Sonoma (14.4) with Anaconda3 dev. env.|LW-klayout-0.29.0-macOS-Sonoma-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(33)|
34|Sonoma (14.4)|HW-klayout-0.29.0-macOS-Sonoma-1-qt6MP-RsysPhb311.dmg|MacPorts|OS-bundled|Homebrew|(34)|

(30) shares OS-bundled Ruby2.6 and Python3.9 (dev. tools required); Qt5 from MacPorts is embedded.
(31) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(32) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(33) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(34) **experimental**; shares OS-bundled Ruby2.6; Qt6 from MacPorts and Python-3.11 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Big Sur (11.7.10)|From 0.28.13 onward, DMGs will not be provided for this OS.||||(1p)|
7|Catalina (10.15.7)|From 0.28.13 onward, DMGs will not be provided for this OS.||||(1p)|
8|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
10|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
11|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) From 0.28.13 onward, build is no longer possible.

With warm regards,
Kazzz-S

klayoutmatthias, 2024-04-06:
Thanks at lot! Files are on the server and I will leave this ticket open.

I did not dare to publish the Qt6 binaries as I feel that would create a support issue. Let me know if you think it makes send to publish them too.

Regarding Qt6: I am on Ubuntu 22 and with the Qt 6.4.2. 2.5d view is working on my combination (Nvidia T1200 with version 550 driver):

[image]

[image]

However, OpenGL is highly dependent on the driver and on the Mesa library, so that is just a personal snapshot.

Apart from that: when you say some Qt-based Python macros do not run - is that because of the changed API (Qt6 vs. Qt5), or is that not related to that? 

Best regards,

Matthias

Kazzz-S, 2024-04-06:
Hello @klayoutmatthias,

 > I did not dare to publish the Qt6 binaries as I feel that would create a support issue. Let me know if you think it makes send to publish them too.

Thanks for your consideration.
I agree with you. It's a wise decision. Let's see the response from the community.

> Apart from that: when you say some Qt-based Python macros do not run - is that because of the changed API (Qt6 vs. Qt5), or is that not related to that?

I'm not entirely certain so far. However, I feel this is most likely due to the changed API.

### Example-1 (KLayout's example) ###
[image]
[image]

### Example-2 (my own) ###
[image]
The above dialog (I hope the function is self-explanatory) does not show up in the Qt6 version.

In the console, I got the warning:
[image]
In the source, I have written:
[image]

Best regards,
Kazzz-S

Kazzz-S, 2024-04-07:
cont.

>> Apart from that: when you say some Qt-based Python macros do not run - is that because of the changed API (Qt6 vs. Qt5), or is that not related to that?

> I'm not entirely certain so far. However, I feel this is most likely due to the changed API.

It looks like both the changed API and other causes are mixed up.

(1) I have changed the code to fix the above warning.
[image]

(2) Then another warning comes up.
[image]

(3) In the Qt5 version binaries:
[image]

(4) But in the Qt6 version binaries:
[image]

I'll check this in my office Linux environment tomorrow.

Regards,
Kazzz-S

klayoutmatthias, 2024-04-07:
Hello @Kazzz-S,

The QFont object got some more constructors in Qt6, so there may be some ambiguity. I can't see how exactly you call QFont, but this version is working both on Qt5 and Qt6:

```
pya.QFont("helvetica", 12)
```

(12 is the point size). I suspect that the first argument is not exactly a string. In Qt6, you can also give multiple families, maybe that creates the ambiguity:

```
# New in Qt6
pya.QFont(["helvetica", "times"], 12)
```

(1) Regarding "QIODevice": Using "QIODevice.ReadMode" definitely is universal. They have split "QIODevice" into "QIODevice" and "QIODeviceBase" and "OpenModeFlag" now is inside "QIODeviceBase" as you correctly noticed. 

(2) Indentation errors usually come from Python - maybe some Python versions are more picky then others. I don't think this is related to the Qt version.

(3) The "QtDesigner" module was not really a good idea from the beginning. It is made for creating Qt Designer plugins, but I provided it mainly for the "QFormBuilder" class which can be replaced easily by "QUiLoader" from the "QtUiTools" module. The latter is available since many years now and is the better alternative in terms of functionality and support within Qt distributions. 

Do you see a specific need for the "QFormBuilder" class?

Thanks and best regards,

Matthias

Kazzz-S, 2024-04-07:
Hello @klayoutmatthias,

Thanks for these explanations.
I'll study one by one, taking time.

> Do you see a specific need for the "QFormBuilder" class?

The answer is YES and NO.
When I started studying Python macros with UI, the sample program was a golden reference. 
[image]

Since it worked fine, I never considered changing the code until today. 
But I do not remember where I picked `QIODevice_OpenModeFlag.ReadOnly` from;-(  
Maybe from some Web info.

Basically, I do not need to create Qt Designer plugins.
So I can refactor the code using "QUiLoader".
This will be my work today. :-)

With best regards,
Kazzz-S

Kazzz-S, 2024-04-08:
cont.

### The Font issues ###

The problem was the usage of the **font-weight integer constant.**
By changing the code, as shown below, the dialog appears as (almost) expected.
[image]
[image]
[image]

Thanks for the hint.
Kazzz-S

Kazzz-S, 2024-04-08:
cont.

### The 2.5d issues ###

**About GPU**
[image]

**In the Qt5 environment**
[image]

**In the Qt6 environment (using the debug build)**
[image]

I'm unsure how much it would be helpful, but I attached the Valgrind output.
[2.5D-logs.zip](https://github.com/KLayout/klayout/files/14901891/2.5D-logs.zip)

Best regards,
Kazzz-S

lukasc-ubc, 2024-04-14:
Hello @Kazzz-S @klayoutmatthias 

I notice that we are missing a HW version for this release on the KLayout web page.  Kazz created a Qt6 HW but not Qt5.  Would it be possible to have this HW version available?  My students and I rely on Python3, and having it embedded makes easy to install and use.  I'd rather not supporting people having to install their own Python3 environment and using the LW version.

I tried the ST version, and on Monterey it does show up with Python 3.  But I get a lot of access messages (despite having allowed them multiple times, and approved them in System Settings > Privacy > Files and Folders.

[image]
[image]

I then installed 0.27.19 ST Monterey version, and I do not get these access problems.  So it seems like something in 0.29 is causing an issue where it keeps asking for access permission.

Strangely, I was able to solve fix it, by manually saving a layout to the folders it was asking about.  Then the approval seemed to stick.

So, ST vs HW -- is there a reason for me to ask for HW version, or is the ST one fine?

thank you

klayoutmatthias, 2024-04-16:
@Kazzz-S Many thanks for the new DMGs. Let me know if they do the job and I will place them on the server.

Best regards,

Matthias

lukasc-ubc, 2024-04-16:
This version works for me:  HW-klayout-0.29.0-macOS-Sonoma-1-qt5MP-RsysPhb311

Thank you

Kazzz-S, 2024-04-16:
Hello @klayoutmatthias,

Please place the new three DMGs uploaded [here](https://www.dropbox.com/scl/fo/nylhe3txbl58r3ct2r6d8/AMRysYvgQgaQPgdmBQ2HtjE?rlkey=xex10cjcolaaht7v6ekgbx6xh&dl=0) on the server.

I'll build the `HW-*.dmg` from the next maintenance release using the (qt, ruby, python)=(qt5macports, sys, hb311) parameter set.
I'll also update the contents of the `Homebrew-HUser-ReadMeFirst.txt` accordingly.

Best regards,
Kazzz-S

klayoutmatthias, 2024-04-17:
Dear @Kazzz-S  and @lukasc-ubc,

Thanks @Kazzz-S for the update! 

I have placed the files on the server. I invented this description, maybe you can check if that makes sense:

[image]

Thanks, and best regards,

Matthias

Kazzz-S, 2024-04-17:
Hello @klayoutmatthias,

Thank you for the new description. It's concise, and I like it.

Best regards,
Kazzz-S

klayoutmatthias, 2024-05-06:
Closed together with #1702. 

Thanks a lot! :)