KLayout/klayout #1678: Qt major version-wise Assistant

Kazzz-S, 2024-04-07:
This is a feature request.
There are API differences between Qt5 and Qt6. It would be appropriate to generate `$HOME/.klayout/help-index.xml` Qt major version-wise.
For example,
* `$HOME/.klayout/help-index-qt5.xml` for Qt5
* `$HOME/.klayout/help-index-qt6.xml` for Qt6

I think this kind of treatment will make the transition from Qt5 to Qt6 easier.

Best regards,
Kazzz-S

klayoutmatthias, 2024-04-07:
Hi @Kazzz-S,

thank you for this proposal. 

I think this makes it easier if you switch between the Qt versions, right? Normally, every KLayout installation should write the help index according to the Qt version it was built with.

I am considering writing the help index with KLayout's version too. Plus maybe compressing it as it is already quite large. So like "help-index-0.28.17-qt5.xml.gz".

Best regards,

Matthias

Kazzz-S, 2024-04-07:
Hi @klayoutmatthias,

> I think this makes it easier if you switch between the Qt versions, right? 

Yes.

> Normally, every KLayout installation should write the help index according to the Qt version it was built with.
> 
> I am considering writing the help index with KLayout's version too. Plus maybe compressing it as it is already quite large. So like "help-index-0.28.17-qt5.xml.gz".

I think this is a good idea.

Best regards,
Kazzz-S

closed 2024-04-17 by klayoutmatthias

Kazzz-S, 2024-04-20:
Hi @klayoutmatthias,

Thank you for implementing the new feature, which I have successfully verified on Linux. 

Best regards, 
Kazzz-S

klayoutmatthias, 2024-04-20:
Very good. Thank you for your feedback! :)

Matthias