KLayout/klayout #1680: [Qt6] QRegExpValidator class

Kazzz-S, 2024-04-08:
This is an enhancement request regarding migration to Qt6.

The class 'QRegExpValidator' is not included in PYA so far.
[image]

I would also like to have it in the Qt6 environment since I use it to validate an IP address, as shown below.

https://github.com/KLayout/klayout/assets/16847255/ebe2c5c0-0310-4aea-b619-8c848472f053

Best regards,
Kazzz-S

klayoutmatthias, 2024-04-10:
Hi @Kazzz-S,

Qt added `QRegularExpressionValidator` in Qt5 (leaving `QRegExpValidator`), but removed `QRegExpValidator` in Qt6. 

I'm afraid you will need to migrate to `QRegularExpressionValidator` for Qt6 compatibility. My understanding is that both are pretty much the same.

Best regards,

Matthias

Kazzz-S, 2024-04-10:
Hello @klayoutmatthias,

Thanks for the info.
I should have searched the assistant document more carefully.
With the following changes, the tool works fine in both Qt5 and Qt6 environments.
[image]

Best regards,
Kazzz-S

closed 2024-04-10 by Kazzz-S