KLayout/klayout #1688: Question about available KLayout versions in pypi

qpavsmi, 2024-04-16:
I have a question about the lifecycle policy of older standalone KLayout releases. Currently in pypi (https://pypi.org/project/klayout/#history) there is a limited number of versions listed, but I seem to recall that we used to be able to fetch version `0.28.1` from pypi. It looks like there is a policy in place here and just wanted to clarify that I got it right: are you keeping only the latest release for each major version (hence `0.26.12`, `0.27.13`, `0.28.17.post1`) before the most recent major version, and then keeping each release within the current major version?

The reason I ask is because we have had some regression test jobs in our CI to run tests for specified older KLayout versions to be aware if our code development breaks anything for older KLayout versions. The CI would pull the KLayout packages from pypi, and I got confused that it stopped pulling version `0.28.1` specifically.

klayoutmatthias, 2024-04-16:
Hi @qpavsmi,

Thanks for asking and sorry if I caused some inconvenience.

The problem is basically the limited space on PyPI project accounts. I had to remove older versions to make space for the new 0.29.0. Unfortunately the binary releases are pretty heavy and the Python/OS matrix is large, so there is space for about 20 versions on PyPI. I have tried to explain this here: https://www.klayout.de/forum/discussion/2485/old-versions-gone-on-pypi. I can basically reduce the Python version matrix to reduce the disk space needed per release. Some advice on must-have Python versions is appreciated.

I was trying to retain the latest release of a major branch (0.28.17 in that case). I am assuming it is a good enough representative for the 0.28.x branch. For 0.29.x which I want to retain all minor version until I switch to 0.30.

Does that make sense?

Best regards,

Matthias

qpavsmi, 2024-04-17:
Thanks, make sense given the circumstances! I'll keep this in mind as I configure our CI processes. I'm not sure myself why we specifically nailed down 0.28.1 version for regression checking (some historical reason that no one remembers anymore), so I think we can just configure CI to use the available versions. No need to add other intermediary KLayout releases on our account.

Best,
-Pavel