KLayout/klayout #1700: Does it help? 2.5d view on OpenGL 3.0

klayoutmatthias, 2024-05-02:
In order to deal with a number of reports I received with respect to non-functional 2.5d views, I ported the view to GLSL 1.20 and removed the geometry shader. It should be compatible with OpenGL 3.0 now.

I have some hope this fixes the issue on MacOS.

The code is in the `d25view-fixing` branch (https://github.com/KLayout/klayout/tree/d25view-fixing). 

It could use some memory optimization, but first I would like to have some feedback. 

If you want to help, please give it a try and let me know if your 2.5d problem is fixed.

Thanks and best regards,

Matthias

klayoutmatthias, 2024-05-02:
No, you don't :)

Looks like there is a bug in my shader program. Interestingly it compiled on Linux.

Matthias

Kazzz-S, 2024-05-03:
Hello @klayoutmatthias,

Thanks a lot for trying to fix this issue.
I built the `d25view-fixing` branch on **macOS Monterey** (Intel Mac).

First, I got a syntax error relating to ";" but you have already fixed it.
[image]

Next, I got some compilation errors at runtime.
[image]

I've modified the code as below.
[image]

Then, I could successfully get the 2.5D view. (Also refer to #1569)

https://github.com/KLayout/klayout/assets/16847255/39d59f8b-72e0-406f-8252-b7d6996c709a

Best regards,
Kazzz-S

Kazzz-S, 2024-05-03:
cont.

The above application bundle built on `Monterey` can also run on `Ventura` and `Sonoma`. 
[image]

[image]

Kazzz-S

Kazzz-S, 2024-05-03:
## 2. Linux Mint 21.3 ##

Using the same source code as macOS...
[image]
[image]

Kazzz-S

stefanottili, 2024-05-03:
Works on M1 macOS Sonoma too.

klayoutmatthias, 2024-05-03:
Very good. Thanks for the patch (interesting, how different the GLSL implementations are) and the feedback!

I am happy to hear it works on MacOS too - OpenGL really is weird. 

I will include the new version in the next 0.29.1 release (tomorrow, hopefully). From the memory perspective, I conducted an experiment with about 20M shapes (boxes). During 2.5d rendering with some booleans and mesh lines, memory goes up to 14G, but I hope that is an extreme case. As I was removing the geometry renderer, I now need to supply normal vectors for the lighting and this doubles the vertex memory. It is possible to reduce the footprint by using indexed buffers, but I first need to learn how to use them properly. I spare that for a future release :)

Best regards and thanks again,

Matthias

klayoutmatthias, 2024-05-03:
My 20M rectangle test case (sluggish ...)

[image]

ejprinz, 2024-05-10:
Tested on RedHat EL8. Works fine.

klayoutmatthias, 2024-05-13:
Closing, as it is released. Thanks for your support!

closed 2024-05-13 by klayoutmatthias