KLayout/klayout #1704: Registered github Salt package using group feature seems to be found in package index

FatsieFS, 2024-05-07:
I registered a KLayout package on URL git+https://github.com/FatsieFS/Salt-C4M_FlexLibs.git.  
It is using the group feature as allowed inside KLayout when creating new package from managing packages; package name is `C4M/FlexLibs`.  
After having used the confirmation email for the package, the package does not seem to be listed in the Index nor it is installable from the salt package manager inside KLayout.

klayoutmatthias, 2024-05-10:
The problem is not related to the groups - I found the package was not confirmed, hence not visible. I changed the status to ACTIVE and now it is visible in the package list.

Did something happen while confirming the package? I see the mail went out and the a request was issued with an URL like "/https:/sami.klayout.org/register_confirm?id=...", instead of "https://sami.klayout.org/register_confirm?id=...". So I assume something went wrong while clicking on the confirmation link?

Best regards,

Matthias

FatsieFS, 2024-05-10:
OK, good. Did not notice a problem with confirmation and when following the confirmation link again it said the URL was not existing.
Anyway I am going to register three more packages with grouping so will let you know if everything goes OK this time.

klayoutmatthias, 2024-05-11:
Well, maybe some glitch on the server ... I have not had similar reports so far.

Let me know if you encounter the problem. If not, we can close this ticket.

Best regards,

Matthias

FatsieFS, 2024-05-13:
In now registered succesfully three Chips4Makers PDKs under C4M/PDK so all is fine.
I may have even copied the wrong link (e.g. the reject link).

closed 2024-05-13 by FatsieFS