KLayout/klayout #1706: [macOS] User Properties Dialog Crashes

Kazzz-S, 2024-05-09:
## 1. The Latest Version (0.29.1) ##

The `User Properties` dialog crashes when I click the `[User Properties]` button.
This problem reproduces with the latest four binaries listed below.
[image]
[02901-crash-logs.zip](https://github.com/KLayout/klayout/files/15268710/02901-crash-logs.zip)

I used this feature before. I recently started using it again after a long hiatus and noticed this problem.

This problem is macOS-specific.
This feature works as expected in the Linux and Windows environments, as shown in the images below.
[image]
[image]

To be continued...

Kazzz-S, 2024-05-10:
Cont.

## 2. Previous Versions (< 0.29.1) ##

I tested previous versions with HW*.dmg going back in history.
Then, I found versions up to 0.28.8 are problematic.
```
<0.29.1> --> 0.28.17 --> 0.28.15 --> 0.28.14 --> 0.28.13 --> 0.28.12 --> 0.28.11 --> 0.28.10 --> 0.28.9 --> 0.28.8
```
[previous-crash-logs.zip](https://github.com/KLayout/klayout/files/15268879/previous-crash-logs.zip)

To be continued...

Kazzz-S, 2024-05-10:
Cont.

## 3. Further Previous Versions (< 0.28.8) ##
I tested with `0.28.7`, `0.27.13`, and `0.26.12` and found they are OK, as shown in the sample images below.
[image]
[image]
[OK-samples.zip](https://github.com/KLayout/klayout/files/15268945/OK-samples.zip)

I suspect some problem may have intruded between `0.28.7` and `0.28.8`.

Best regards,
Kazzz-S

klayoutmatthias, 2024-05-10:
Thanks for the analysis. I will look into that.

Matthias

Kazzz-S, 2024-05-11:
Cont.

## 4. With Qt6 ##

It works with Qt6!

[image]
----
[image]

Kazzz-S, 2024-05-11:
Cont.

## 5. Qt-wise Status ##
[image]

[Qts.xlsx](https://github.com/KLayout/klayout/files/15281954/Qts.xlsx)

Thanks,
Kazzz-S

stefanottili, 2024-05-12:
on M1 lastest master:
homebrew qt 5.15.13 coredump
homebrew qt 6.7 works

klayoutmatthias, 2024-05-12:
Dear @Kazzz-S  and @stefanottili

Thanks for the many datapoints and your experiments!

Maybe I found the root cause: on some platforms / compilers, a default-initialized iterator appears no to have the same value.

My patch is here: https://github.com/KLayout/klayout/pull/1708

Maybe you can try whether this fixes the problem.

Thanks and best regards,

Matthias

klayoutmatthias, 2024-05-12:
P.S: I hope you did not spend too much effort in your research. Basically all I needed was the hint about 0.28.7 vs. 0.28.8 :(

stefanottili, 2024-05-12:
git checkout origin/bugfix/issue-1706 fixes the coredump on M1 Sonoma.

The fact that it went unreported for so many versions just might indicate that nobody is using this feature ...

Kazzz-S, 2024-05-13:
Hello @klayoutmatthias,

Thank you for trying to fix this problem.
Using the patch #1708, I rebuilt:
1. Four binaries with Qt5.x mentioned in [this table](https://github.com/KLayout/klayout/issues/1706#issue-2288649465).
2. Two binaries with Qt6.x mentioned in [this comment](https://github.com/KLayout/klayout/issues/1706#issuecomment-2105605659).
3. One binary with Qt5.12.8 on Linux Mint 20.3.

I confirmed that all 7 test cases are OK.

Thank you again, and best regards,
Kazzz-S

Kazzz-S, 2024-05-13:
Hello @stefanottili,

Thank you for testing on M1 Sonoma.
I use this feature mainly in Linux environments, so I overlooked this problem till recently.

Best regards,
Kazzz-S

closed 2024-05-31 by klayoutmatthias