KLayout/klayout #1710: feature request: create separate cell for def FILL

stefanottili, 2024-05-15:
Some fallout of the discussion about FILL with regard to openroad.

https://github.com/The-OpenROAD-Project/OpenROAD/issues/5114

Looking at the filled gds, it would be nice to have a separate hierarchy (ref + cell) for the def FILL info.
It could be named <designname>_BE_FILL and referenced at 0,0 or the lower left of the DIEAREA if 0,0 lies outside of it.
The latter case would require the FILL coordinates the be shifted accordingly.

klayoutmatthias, 2024-05-16:
Some "fallout" that is ... :)

I am trying to understand how this is useful in that case. Just placing Fill on a different layer is not helping?

Matthias

stefanottili, 2024-05-16:
Having fill in a separate cell has two benefits, the first being the more important one.

a) ease of “hide fill to look at the other things”.

Looking at a layout with fill, I use the “hide cell” by double clicking on the fill cell in the hierarchy browser.
Much less clicking then hiding the individual fill layers in the layer browser.
But this requires the fill to be in a separate cell from the place and route data.

If you use physical verification tools to generate fill, that’s how you get it.
But not if the metal fill is generated by OpenRoad and be part of the def.
Which makes it easier for the cap extraction ..
One still needs a pv tool to generate fill for inside leafcells and for front end layers.
 
 b) ease of deleting fill from a layout.

E.g. to use a different tool to create fill or for eco’s.

closed 2024-05-31 by klayoutmatthias