KLayout/klayout #1714: Intermittent <defunct> cell name change

lukasc-ubc, 2024-05-20:
Hi @klayoutmatthias

I have a strange bug:

Start with the attached GDS, which was created using the technology "EBeam", and view the cell listing:
[image]

Open the same file without the Technology available in KLayout:
[image]

Save as (without Context info), then load again, results in:
[image]

Notice that the cell <EBeam.ebeam_gc_te1550> is instantiated in two cells, and after saving, one of them is renamed "PhC14".  I have no idea where this name is coming from!

I came across this problem when running it in a Python script.  I was loading the file, without a technology in place, and noticed that cell.name was PhC14, whereas cell.basic_name() was correctly given as ebeam_gc_te1550.

    # Load layout  
    cv = pya.MainWindow().instance().load_layout(f)
    layout2 = cv.layout()
    
    # Debugging
    print('*** Cell names in layout %s, MainWindow load' % f)
    for c in layout2.each_cell():
        print(c.name)
        print(c.basic_name())

gives the following:

```
ebeam_gc_te1550 ebeam_gc_te1550
TEXT TEXT
TEXT$1 TEXT
PhC14 ebeam_gc_te1550
PBSD3TE PBSD3TE
gc3 gc3
EBeam_openEBL_arunaPBS3 EBeam_openEBL_arunaPBS3

```

Any idea how this is happening?

Strangely, out of fifty files, only this one has this problem.

This is with KLayout 0.29.0 and 0.29.1

thank you
Lukas

[EBeam_openEBL_arunaPBS3_LC.gds.zip](https://github.com/KLayout/klayout/files/15374010/EBeam_openEBL_arunaPBS3_LC.gds.zip)

lukasc-ubc, 2024-05-21:
Hi Matthias,

I exported the GDS into the text format, and that helped provide more details.  The student that generated the design indeed created a GDS in another tool, saved it as PhC14.gds, then edited the layout in KLayout.  So somehow the name got attached.

```
SNAME PhC14
XY 0: 0
PROPATTR 1 
PROPVALUE CELL=ebeam_gc_te1550
PROPATTR 0 
PROPVALUE LIB=EBeam
ENDEL 
SREF 
SNAME ebeam_gc_te1550
XY 0: 0
PROPATTR 1 
PROPVALUE CELL=ebeam_gc_te1550
PROPATTR 0 
PROPVALUE LIB=EBeam
ENDEL 
ENDSTR 
```

I wasn't aware that the SNAME (basic_name) property could be different than then PROPVALUE CELL.  Indeed, doing the Rename operation shows the SNAME (PhC14).

I added this check in case it happens again:

```
    for c in layout.each_cell():
        if c.basic_name() not in c.name:
            raise Exception ('- WARNING: cell SNAME (basic_name = %s) does not match the cell name (name = %s). See https://github.com/KLayout/klayout/issues/1714' % (c.name, c.basic_name()) )
```

On a related note, I was just checking the cell properties to see if I can see these values.  I do an Edit > Properties, then User Properties, and I get a KLayout crash.  I notice it was already reported and fixed: https://github.com/KLayout/klayout/issues/1706

thank you
Lukas

klayoutmatthias, 2024-05-21:
Hi Lukas,

yes, the crash-on-show properties bug is fixed. It only happens on MacOS, that's why I did not catch it. I will release 0.29.2 end of this month.

Best regards,

Matthias

P.S: Okay to close this issue, or should we leave it open until there is a clue how the two cells were created?

closed 2024-05-21 by lukasc-ubc