KLayout/klayout #1716: Klayout not recognizing hatches made in dxf files in AUTOCAD

aryaman3900, 2024-05-20:
I have to use maskless aligner for fabricating a microfluidic device for which I need to have my channels filled with something. So I used hatch option to create hatches within the channels as shown in this image:

[image]

But when I am opening the same file in Klayout to convert to GDSII some additional lines are been shown in the drawing and also the circles are filled which shouldn't be. Can someone please tell how to fix this issue?

[image]

Kazzz-S, 2024-05-21:
Hello @aryaman3900,

I hope my comments can help you.

### 1. Sample Data ###
* [samples.zip](https://github.com/KLayout/klayout/files/15384755/samples.zip)
[image]
[image]

Designed by usisng:
[image]

to be continued...

Kazzz-S, 2024-05-21:
Cont.

### 2. Reader Options ###
[image]

Kazzz-S, 2024-05-21:
Cont.

### 3. Without Hatching ###
[image]
[image]
[image]
[image]
[image]

Kazzz-S, 2024-05-21:
Cont.

### 4. With Hatching ###
[image]
[image]
[image]
[image]
[image]

Regards,
Kazzz-S

klayoutmatthias, 2024-05-24:
Hi @Kazzz-S,

thanks a lot for your experiments and insights.

DXF to GDS is not a simple topic. DXF has too many flavors. Hatches are converted to polygons in KLayout, but they may overlay the polygons created by the POLYLINE objects. So I'd not use them.

Usually, only lines are drawn and the filling is generated automatically by KLayout. 

Matthias