KLayout/klayout #1722: [macOS] Crash when opening layout having Custom Macro Menus

martinjankoehler, 2024-05-27:
@klayoutmatthias this is the same issue as with the IHP LVS menu

Let's say I create a macro, using top-level menu "MJK":
[image]

[image]

Then if I open a layout, the crash happens.
[image]

klayoutmatthias, 2024-05-28:
Hi Martin,

I tried to reproduce this problem on Linux without success (yet). I ran the process through valgrind which usually is a good way to detect real issue, but no error was reported.

The stack trace unfortunately just points to Qt internals and there is nothing which looks wrong in particular. Maybe there is something with the QAction objects that KLayout generates, but there is no hint what.

Just to confirm: I have attached two .lym files:

[issue_1722.zip](https://github.com/KLayout/klayout/files/15474752/issue_1722.zip)

So if you prepare a new KLayout home folder, like:

```
mkdir ~/.klayout-test
cd ~/.klayout-test
mkdir pymacros
cd pymacros
unzip /path/to/issue_1722.zip
```

and start KLayout with this isolated environment using:

```
KLAYOUT_HOME=~/.klayout-test klayout
```

then you should see a new top level menu "MJK" and two entries (Menu1, Menu2). You mean that now, if you load any layout, the program crashes?

Matthias

Kazzz-S, 2024-05-28:
Hi @klayoutmatthias,

I'm testing this case.
Thanks for your instructions. 
[image]
The problem is reproduced on **macOS Monterey (Intel Mac)**.
I'll upload the results later.

Kazzz-S

Kazzz-S, 2024-05-28:
### 1. KLayout 0.29.1 on macOS Monterey (Intel Mac) ###

* The custom macros tested: [macros_KazzzS_1722.zip](https://github.com/KLayout/klayout/files/15476899/macros_KazzzS_1722.zip)

1. File --> Open --> [Finder] --> File Selection --> Open ==> [Crash Log](https://github.com/KLayout/klayout/files/15476901/klayout_crash-ST.zip)
2. File --> Open Recent --> File Selection ==> Crash
3. [Finder] --> File Selection --> Double Click ==> No Crash (macros works)

https://github.com/KLayout/klayout/assets/16847255/ff08343f-d639-487a-adca-444fa957f3cf

To be continued.

Kazzz-S, 2024-05-28:
Cont.

### 2. Linux and Windows10 ###

* The macros work normally.

**Linux Mint 20.3**

https://github.com/KLayout/klayout/assets/16847255/6c6b8295-bd8d-4935-a7e2-9add30bf574a

**Windows10**

https://github.com/KLayout/klayout/assets/16847255/5f79098a-a7c4-48fb-bc83-8bc6712c75e2

Kazzz-S, 2024-05-28:
### 3. Previous macOS versions ###

* See the table below.
* There seem to be two boundaries: (a) between 0.28.17 and 0.28.16 (b) between 0.28.4 and 0.28.3.
[image]
[Qts.xlsx](https://github.com/KLayout/klayout/files/15476948/Qts.xlsx)

Kazzz-S, 2024-05-28:
### 4. Placing the macros at different locations ###

* The three cases blow work normally with KLyout 0.29.1 on macOS Monterey (Intel Mac).
* It looks like the problem appears when the custom macros menu is placed at the very last (rightmost) of the menu bar.

[image]
[image]
[image]

Regards,
Kazzz-S

klayoutmatthias, 2024-05-31:
Dear all (and in particular @Kazzz-S  :) ),

The problem is actually a known one ... we have this case a number of times already. I thought I could work around it, but that does not seem to work.

There is a general problem, that Qt apparently cannot dynamically modify the top level menu on MacOS. It is possible to add items, but not to remove / replace them. As menus can be technology specific, loading a file tries to re-generate the menus and this will crash the application.

I got myself a Mac M1 / Sonoma cloud instance on Scaleway to debug the problem. It is somewhat painful (+slow), but possible. I found that on Qt 6.7.0, the application no longer crashes, but the "MJK" menu is shown twice. Not much better ..

I could not really debug into the Qt code responsible for this. My current approach now is to handle dynamic top level menus differently on MacOS: I will try to introduce a new top level menu, i.e. "Extra" and collect all dynamic top level menus there as sub-menus. This means that the top level menu will look somewhat different on MacOS, but the application should not crash.

Keep you updated.

Matthias

klayoutmatthias, 2024-05-31:
The solution I came up with now looks like this:

[image]

Would that be acceptable?

Matthias

Kazzz-S, 2024-05-31:
Hello @klayoutmatthias,

Thank you for fixing this problem.
The proposed solution is intuitive and looks smart.
I did not modify the existing *.lym files, but they were automatically placed in the `...` group.
Can I still change the order of sub-menus in the group?
I hope the "Macros in Menus" documentation will be updated accordingly.

https://github.com/KLayout/klayout/assets/16847255/a97d0bcf-98e9-4df9-8603-d0a0f5d221bd

Best regards,
Kazzz-S

martinjankoehler, 2024-05-31:
Many thanks for testing @Kazzz-S 
Thanks for the solution @klayoutmatthias
Let me try to debug this weekend, perhaps we get it working 1:1 as in Linux.

klayoutmatthias, 2024-05-31:
Hi @Kazzz-S,

thanks for the hint about the documentation. I added a brief note:

[image]

Regarding the order: the idea is to shuffle all non-standard main menu items and place them in the "..." menu in the same order they would appear on the main menu, i.e. on Windows or Linux.

For example: if you define two custom, top-level menus, say "A" and "B" with:

* "A" placed before "tools_menu"
* "B" placed at ".end" (the end of the main menu)

Then these menus will not appear in the main menu, but in "...". The order will be "A", then "B", because "A" would be placed before (left of) "B". On the positive side, the application should not crash :)

If you now define a third menu, say "C" and configure it "before B", then the order in the "..." menu would be: "A", "C", "B".

Matthias

Kazzz-S, 2024-06-01:
Hello @klayoutmatthias,

Many thanks for your explanations.

Meanwhile when conducting the QAT for c148f591, I got a signal segmentation violation error below.
Do you think this is a side effect of the changes going on?
The attached ZIP contains six full log files. Out of six, five show the same error.

Kazzz-S

```
:
:
:
    ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
    1140.25 tests/s, 55872.29 assertions/s
Time: 0.081s (wall) 0.07s (user) 0s (sys)
Memory: 0k
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Running rbaTests:qtbinding
    Loaded suite QtBinding_TestClass
    Started
    ..........Signal number: 11
    Address: 0x0
    Program Version: KLayout 0.29.1 (2024-06-01 rc148f591b)

    Backtrace:
    /Users/sekigawa/GitWork/klayout/qt5MP.build.macos-Monterey-release-RsysPsys.macQAT/libklayout_lay.0.29.1.dylib +0x20e32b _ZN3lay14signal_handlerEiP9__siginfoPv
    /usr/lib/system/libsystem_platform.dylib +0x3dfd _sigtramp
    /usr/lib/libobjc.A.dylib +0x1d0ba _ZN7cache_t6insertEP13objc_selectorPFvvEP11objc_object
    /opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x115868 _ZN8QMenuBar11actionEventEP12QActionEvent
    /opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0x347b1 _ZN7QWidget5eventEP6QEvent
    /opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0xd748 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
    /opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets +0xe5a2 _ZN12QApplication6notifyEP7QObjectP6QEvent
    /Users/sekigawa/GitWork/klayout/qt5MP.build.macos-Monterey-release-RsysPsys.macQAT/libklayout_lay.0.29.1.dylib +0x7b6c5 _ZN3lay14GuiApplication9do_notifyEP7QObjectP6QEvent
    /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore +0x135852 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
:
:
:
```

[ut_runner-c148f591.zip](https://github.com/user-attachments/files/15520662/ut_runner-c148f591.zip)

klayoutmatthias, 2024-06-01:
Hello @Kazzz-S,

thanks for reporting this. I was able to reproduce the problem and I think I fixed it. There was another hidden path, by which top level menu items got removed, hence triggering the same crash.

I have updated the bugfix/issue-1722 branch. On the M1 Sonoma instance I booked, now most tests pass or at least execute without problems. I needed to exclude three tests because of the persistent XML-throws-Exceptions-which-are-not-caught problem. 

Matthias

Kazzz-S, 2024-06-01:
Hello @klayoutmatthias,

I re-ran QAT on commit 4b0a3dff and confirmed these issues no longer occur.

Kazzz-S

klayoutmatthias, 2024-06-02:
Hi @Kazzz-S,

having the opportunity to run experiments on a M1 Mac, I added some more commits to the branch:
* fixing the XML parser issue - I avoid all exceptions now and use the QXmlContentHandler features to control errors. The three (or four) tests now should run without crashing and do not need to be excluded any longer.
* providing changes in the Qt binding files for Qt 6.7 compatibility - this is the version I found on Homebrew and it lacks a few classes and methods
* updating some golden test data with new variants for MacOS. This is required as some implementations (notably hierarchical DRC and LVS) uses std::hash for performance reasons and the results are reproducible, but implementation-dependent.

As a result, only two tests are failing now on the M1/Sonoma/Homebrew/Qt6.7 combination. These are tests that spawn many threads under heavy load and are not necessarily meaningful on a different architecture.

I hope this solves some issues that existed since long now (specifically the XML parser issue).

I will not keep the M1 cloud instance for long - it is awful to work with, it is hard to get one and it is somewhat expensive over a longer period. I hope I can archive the data there and repeat this action once again if needed.

Best regards,

Matthias

Kazzz-S, 2024-06-03:
Hello @klayoutmatthias,

Thank you very much for improving the QAT environment for macOS.
I have tested the 23b74339 commit and confirmed that the three (or four) tests now could run without crashing and no longer need to be excluded.

The attached ZIP contains six (* see below) log files collected on Monterey (Intel Mac).

* 1 x Qt5.15.2 from Anaconda3
* 3 x Qt5.15.12 from MacPorts
* 2 x Qt 6.4.3 from MacPorts

Best regards,
Kazzz-S

[ut_runner-23b74339.zip](https://github.com/user-attachments/files/15531512/ut_runner-23b74339.zip)

stefanottili, 2024-06-03:
yay, M1 homebrew qt 5.15.13, all tests run, no need to exclude anything.

    310 test(s) skipped
    Failed in non-editable mode
        klayout_main_tests:main
        tlThreadedWorkersTests:4
        tlThreadedWorkersTests:5
    Failed in editable mode
        klayout_main_tests:main
        tlThreadedWorkersTests:4
        tlThreadedWorkersTests:5
    6 test(s) failed
Grand total time: 336.685s (wall) 226.45s (user) 52.67s (sys)

klayoutmatthias, 2024-06-03:
Very good .. these are the tests that fail for me too .. "klayout_main_tests" fails because it captures some warning I don't how to suppress. I assume there is some security feature one can disable. In the log files of @Kazzz-S I see some new fails, but only the usual numerical jitters or std::unsorted_xyz effects. All better than a crash and something for the next iteration :)

I'll merge the change then.

Matthias

closed 2024-06-03 by klayoutmatthias