KLayout/klayout #1723: Winget Install

Patarimi, 2024-05-29:
I use winget as a package manager for my CI and I write a (local) manifest for KLayout.
It is available [here](https://github.com/Patarimi/hades/tree/master/manifests/k/KLayout/KLayout/0.29.1]).
If it is useful to others, we could add it to the official [repo](https://github.com/microsoft/winget-pkgs).
What do you think ?

klayoutmatthias, 2024-05-29:
Hi @Patarimi,

Thanks for providing this manifest. I actually did not know it is that easy.

Maybe you could correct the link for the package repo above is it contains the trailing square bracket.

I have no experience with winget myself. What obligations does providing a package come with? To be frank, I am not looking for becoming maintainer of another package. I have enough for myself. Would you take care of that?

Thanks,

Matthias

Patarimi, 2024-05-30:
Hi @klayoutmatthias,

I use winget but I never provide package before. There is an assistant that help with in the creation, deployment and update of the package [wingetcreate](https://learn.microsoft.com/fr-fr/windows/package-manager/package/manifest?tabs=minschema%2Cversion-example). It seems that the only thing to do is update the file with each new version.
If it's just that, I think I can do it. :)

Thanks,