KLayout/klayout #1729: LayerProperties API to create pen/brush/pixmap

EugeneZelenko, 2024-06-04:
Hi, Matthias!

I think will be great to have Qt-specific extension for LayerProperties which will allow to create pen and brush or alternatively create pixmap of specified dimensions based on LayerProperties settings (effective or for isolated layer).

klayoutmatthias, 2024-06-04:
Hi Eugene,

yes, I think I can provide a pixmap. I use a similar function for generating the icons from layer-selection drop-down boxes.
They come with a border reflecting the line style and color. 

Is that what you have in mind? 

I could as well provide the widgets for layer selection boxes. There are some more, like stipple or line style selection widgets.

It's just close to a new release. I don't think I can put this in there already.

Matthias

EugeneZelenko, 2024-06-04:
Hi, Matthias!

Yes, this is exactly what I need!

I'm not sure about widgets, because I'm interesting only in representing final appearance of particular layers, but this may be interesting for other people who would like to customize GUI.

Indeed, this functionality could wait until release would be finished.

klayoutmatthias, 2024-06-04:
Hi Eugene,

there is already such a method :)

Take a look at `LayoutView#icon_for_layer` (https://www.klayout.de/doc-qt5/code/class_LayoutView.html#method115).

Matthias

EugeneZelenko, 2024-06-04:
Hi, Matthias!

Great! Sorry for noise - I should read documentation carefully.

closed 2024-06-04 by EugeneZelenko