KLayout/klayout #1731: sky130A pymacros & gdsfactory

martinjankoehler, 2024-06-06:
Hi @klayoutmatthias,

encounterd some issues trying to enable sky130A pymacros in edit mode
(perhaps not really klayout but rather PDK authors / gdsfactory issues),
but perhaps somebody knows a workaround

## XML parsing issue

The pymacros do not work, due to comment before the XML declaration:
`/foss/pdks/sky130A/libs.tech/klayout/pymacros/sky130.lym`
I guess this issue should really be fixed by the PDK authors, but perhaps klayout could gracefully handle this.

After fixing this manually by moving up the XML declaration to the top, then I get...

## gdsfactory issue

[image]

Are these known issues?

I have those with:
- Docker: hpretl/iic-osic-tools:2024.05
- Docker: hpretl/iic-osic-tools:2024.04 
- MacOS 10.14

Will try to install an older PIP version of gdsfactory.

martinjankoehler, 2024-06-06:
OK, seems with `gdsfactory 6.34.0` they moved `gdsfactory.types` to `gdsfactory.typings`: https://gdsfactory.github.io/gdsfactory/changelog.html#id193

So replacing all the `import` statements gets it to work.

klayoutmatthias, 2024-06-06:
Hi Martin,

I am not sure if I can skip the comment - that is eventually Qt's XML parser. Optional I can use expat, but if the XML is not well-formed, this one would error out is well, I assume.

Matthias

klayoutmatthias, 2024-06-06:
I found this commit: https://github.com/efabless/sky130_klayout_pdk/commit/c49c741c09bf0cb31b9af76aa53f4b9502e0d17a

So apparently the authors are aware of this problem :)

martinjankoehler, 2024-06-06:
Thanks, I‘ll close this issue.

closed 2024-06-06 by martinjankoehler

martinjankoehler, 2024-06-07:
FYI, I've opened an issue there for the import error: https://github.com/efabless/sky130_klayout_pdk/issues/20
and also have a fix handy: https://github.com/martinjankoehler/sky130_klayout_pdk/tree/gdsfactory-types-fix