KLayout/klayout #1734: Clarification re: releaseNotes: 0.29.0 Enhancement: OASIS reader will produce layers that are listed in layer table but empty.

sto6, 2024-06-08:
Questions about the release-notes entry for 0.29.0 : 2024-04-01,

> Enhancement: OASIS reader will produce layers that are listed in layer table but empty.
> Previously, empty layers were never created, even if listed in the layer table.

Q1: Does it mean only the OASIS FILE-FORMAT (but not GDS) can declare/enumerate layers that are never used (i.e. empty, not used in any cell in the OASIS file)?
Q2: It suggests the enhancement may not be backwards compatible as-is: old scripts may behave differently pre-0.29.0 vs 0.29.0 or later?
Q3: If so, from where can I obtain such a demo OASIS file; or how could I make my own demo (via saveAs from interactive klayout with some options to create such an example)?

Below are probably not relevant if answer to Q1,Q2 are no, except perhaps for Q6:

There are many user-forum discussions that in a ruby batch script,
will read a layout from a file (.gds or .oas), and then loop over the layers to process, e.g.:
```
include RBA
ly = Layout.new
ly.read("demo.oas")    # versus: ly.read("demo.gds")
ly.layer_indices.each do |i|
  info = ly.get_info(i)
  puts "doing layer: #{info.layer}/#{info.datatype} ..."
end
```
Q4: When such a `demo.oas` (that declares empty layers) is converted to `demo.gds`, would this script list **more** layers (the empties) **only** in the original oas format, and then **only** in 0.29.0 or later versions?

Q5: If not, what would it take, to convert above script to show/reveal this difference, between .oas vs .gds, and pre 0.29.0 vs post?

Q6: If both oas/gds formats could always declare empties (and the enh. was to correct asymmetry in empty layer treatment in klayout's oas vs gds readers), how should the script detect empties (in order to not output or process them further)?
Thanks for a great tool.

stefanottili, 2024-06-08:
layer/datatype numbers in gds and oasis are associated with either geometry or text.
oasis LAYERNAME are just that, names for layer/datatype (ranges).
There is nothing equivalent to LAYERNAMES in gds.
The oasis spec doesn't include color or stipple information, so it's less than half-baked.
You still need to use tech file when using a viewer to get a consistent view.

klayoutmatthias, 2024-06-08:
Hi @sto6,

Q1: OASIS can store a layer name table. This means it can assign layer names for layer / datatype pairs. While this looks useful at first glance, this table is optional and can be manifold - i.e. assigning multiple names to a single layer. However, if a layer name is present for a layer / datatype pair, KLayout will now create the layer. Without this information, there is no way to tell if a layer is there or not. Only if some shape is present on a layer, the layer truly exists. Some tools, like Calibre from Siemens EDA, write layer names into OASIS, hence it is useful to present this layer, even if it is empty.

Q2: In the new version, you MAY see new layers for the same file, if 1. there is a layer name table and 2. there are empty layers listed in that table. However, if you write your scripts in a way that you propose layers - i.e. using "Layout#layer" with a layer/datatype pair, or DRC scripts with "input", there will be no difference, as these functions created layers on the fly if not present.

Q3: creating such a case is easy: create a new layout, use Edit/New Layer to create a new layer and assign a name in addition to layer/datatype. Save that to OASIS. If you reopen that file with KLayout >0.29.0, you will see an empty layer with the name you gave it.

Q4: GDS will lose layer names and empty layers. They are not listed. Saving an OASIS did not change in 0.29.x. In all versions, saving an empty layer *without a name* in OASIS will not come back, it as no layer table is written. *Named*, empty layers will come back when you *read* them with 0.29 or later.

Q5: If you identify "not present" with "empty" then you have to add an additional check: if you find a layer, you can check it it is not empty, like this:

```
  not_empty = ly.each_cell.find { |c| !c.shapes(li).is_empty? }
```

But in any case, taking the existence of a layer as an indicator for "non empty" is not a valid assumption. A careful analysis of the contents is required, before you can call a layer empty. Here are some reasons why:
* The assumption may not be valid for other or future formats
* A layer may be empty of polygons, but not empty of texts. So the layer is empty in a physical sense, but not in a logical one.
* You may have multiple top cells - while on is empty in terms of a specific layer, the other one is maybe not. So the layer is there, but empty for the first top cell.

Q6: see above.

A general comment: there are basically two use cases for GDS and OASIS: with a stream layer catalog or layer agnostic.

## With a stream layer catalog

This is the case when you have a streamout from a PDK for example. Every tool processing such a layer will have a list of layers it can process. Some layers may be empty, because they are not used. For example, some layer may contains markers for low-voltage devices. This layer will be empty if no such devices are used in your design. A tool processing this layer will need to treat a non-existing layer as empty. So it will create the layer as an empty one, if it does not exist. There is no difference between an empty, existing or non-existing layer. DRC's "input" function works this way for example.

## Layer agnostic

Such tools usually do some kind of analysis. If you see an empty layer, the analysis should turn out something like "nothing to do". For example, if you compute the area density of a layer, an empty layer will give zero. You can chose not to list such layers as irrelevant - hence the difference between a non-existing and existing, but empty layer vanishes. There is a difference, if you expect a certain layer to be non-empty, but then we are back to the case above.

Matthias