KLayout/klayout #1738: Menu item to check for updates

tvt173, 2024-06-12:
hi @klayoutmatthias , I notice that many users download and install klayout and then stay on a stale version for a very long time. i think it would be great to have a `Check for updates...` menu item (maybe under "Help") and possibly a notification within the app when a new version has been released, asking if the user would like to update. i think this would help greatly to keep users on a recent version of the tool

klayoutmatthias, 2024-06-12:
Hi @tvt173,

thanks well for your suggestion. I understand the intention and I know this is a feature present in many other programs. But the idea scares me. An update recommendation comes with a promise, such as "new version is better". 

I don't want to give such a recommendation or warranty. For example, every major release comes with major changes and they may not be mature enough to roll them out everywhere and immediately. I am happy, that this decision is usually taken by third parties, like package maintainers or - in organizations - software pool managers. 

KLayout is still a single-person, spare-time project and detailed management of release quality is somewhat outside my abilities. A release is an offer, nothing more. To me, it is up to the user to accept it or not and that has to be a conscious decision. After all, this is also a potential liability topic.

KLayout shares that model with almost every other tool in the EDA domain or FOSS in general. To bring that to the next level, I'd rather call for some overarching approach. The JKU docker image is very good example for an implementation of that idea (https://github.com/iic-jku/iic-osic-tools).

Matthias

tvt173, 2024-06-12:
Hi @klayoutmatthias , 

thank you for your reply. i understand your perspective but i don't fully agree with it... in the case of klayout, i agree that you shouldn't _nag_ the user to update, but it would be nice to _notify_ them when an update has been made available. for example, I was having an annoying issue with multiple marker browser windows popping up unexpectedly, and when I eventually showed it to one of my colleages he mentioned it was recently patched in 0.29.1 or 2... unless i were to be intently following the changelog, i would not have known this and i probably would have just lived with the annoyance until it bothered me to the point of looking into it further or raising an issue.

some tools will have a pop up on startup for example after a new version has been released, listing the changes and a link to download if you would like. i think that would be a nice level of notification for klayout... not a nagging "please update me" that pops up on every launch, but a "hey FYI we fixed these things" once after each release. 

anyways, understood that it's probably not one of your top priorities, but i thought i'd throw the idea out there as a recommendation. having worked at larger companies too, i've seen many people linger on far outdated versions of klayout not out of choice but complacency. if they were aware of the updates they might choose to do so

Kazzz-S, 2024-06-13:
Hi there,

[KLayout's home page](https://www.klayout.de/) supports [RSS](https://en.wikipedia.org/wiki/RSS).
[image]

I'm using **Chrome** + **Feedbro Reader** to be informed of the recent events.
[image]
[image]

This mechanism is good enough for me.

In my case, however, I monitor the commits to the GitHub repository regularly, so these days, I can predict before the notification that an update is likely to be coming soon :-)

Kazzz-S

tvt173, 2024-06-13:
Thanks for the tip @Kazzz-S !

Yeah it still requires will, so an in-app notification might be more useful for the broader user base. But it's not a huge deal, just a suggestion. Thanks!

klayoutmatthias, 2024-06-13:
Hi @tvt173,

I feel you did not really receive my message. My point is: I don't want users to be mislead by the false impression that a new release is necessarily better. I care for quality, but I can't prevent disasters. I don't want a large number of users following an upgrade advice of mine, and ending up with a non-functional installation. I got enough trouble already.

Managing this is next level. It requires governance. A single-person project does not have this. That is why I call for third parties. Give me an umbrella organization that provides this governance, and an upgrade message will not be a problem. Until then, however, I enjoy happy hacking and leave it to the users to follow me at their own risk.

Matthias

tvt173, 2024-06-13:
hi @klayoutmatthias,

yes, i understand. no hard feelings if you disagree with my suggestion. it's simply a suggestion. thanks for all your great work.

Troy

closed 2024-06-13 by tvt173