KLayout/klayout #1739: Invalid shortcut key

xylewis, 2024-06-13:
When I hit the reset shortcut button
Restart the software, all the shortcut keys are invalid, even Ctrl Z undo ctrl c ctrl v paste copy can not work properly
Is it normal that when you click on the button marked in the picture all the shortcuts won't work even the most basic pasting and copying?
Problem version: 0.28.4
Reuninstall and install 0.26.12, 0.28.4 still has the problem
Installing that version of the software on another computer works fine
[image]

klayoutmatthias, 2024-06-13:
Hi @xylewis,

no, that is not normal. At least that is not intended.

Looks like the "reset" function does a too hard reset and removes the bindkeys. It should better reset them to the defaults.

As a workaround you can delete the configuration file instead of using the "Reset" button. The configuration file is found in 

* ```c:\users\youraccount\KLayout\klayoutrc``` on Windows
* ```~/.klayout/klayoutrc``` on Linux

Deleting this file will reset the configuration entirely.

If you do not want to loose all the other settings, you can delete this line to reset the key bindings:

```
 <key-bindings>...</key-bindings>
```

Thanks for reporting this issue.

Best regards,

Matthias

xylewis, 2024-06-27:
按你指出的第二个方法 
所有快捷键都正常工作了 十分感谢

closed 2024-06-27 by xylewis