KLayout/klayout #1740: batch ruby SIGSEGV due: ly.each_cell.find ... (to find non-empty layers), in 0.27.4 & 0.29.1, libruby.so.2.0

sto6, 2024-06-13:
Trying empty layer code suggested from #1734. `ly.each_cell.find { |c| !c.shapes(li).is_empty? }`
Multiple klayout versions (pre/post 0.29.0) on centos-7.9 (libruby.so.2.0) crash.
The 0.29.1 build used is from your klayout.de, thank you.

As if, just after the first non-empty layer found that way, klayout becomes unstable: crash occurs on next same check (same or another layer), or in exit.

Attached OASIS test-case, klayout script, two logs (script outputs with crash-log). Script ensures that despite differing layer_indexes from the two klayout versions (on same OASIS file), the empty-layer check is done at least once more after a non-empty layer (triggering the crash).

A taste of the logs:

```
% head -30 klay-tst4-0.27.4.log klay-tst4-0.29.1.log
==> klay-tst4-0.27.4.log <==
RUBY_VERSION=2.0.0
indices: [0]
indices: [0]   do_1: false
indices: [0]   do_3: false
test index 0:
... not_empty_0.class: RBA::Cell
test index 0 (again):
ERROR: Signal number: 11
Address: 0x2c
Program Version: KLayout 0.27.4 (2021-10-28 rLatestSourcePackage)

Backtrace:
/apps/klayout/0.27.4/libklayout_lay.so.0 +0x4bbc04 lay::enable_signal_handler_gui(bool) [??:?]
/lib64/libpthread.so.0 +0xf630 __restore_rt [sigaction.c:?]
/lib64/libruby.so.2.0 +0x16d958 rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x171a20 rb_yield [??:?]
/lib64/libruby.so.2.0 +0x5fa5e rb_enum_values_pack [??:?]
/lib64/libruby.so.2.0 +0x171c80 rb_yield [??:?]
/lib64/libruby.so.2.0 +0x69d0b rb_protect [??:?]
/apps/klayout/0.27.4/libklayout_rba.so.0 +0xbc66a std::move_iterator<std::pair<std::string, gsi::MethodBase const*>*>::base() const [??:?]
/apps/klayout/0.27.4/libklayout_rba.so.0 +0x82a53 rba::RubyStackTraceProvider::stack_depth() const [??:?]
/apps/klayout/0.27.4/libklayout_rba.so.0 +0x8369e rba::RubyStackTraceProvider::stack_depth() const [??:?]
/apps/klayout/0.27.4/libklayout_rba.so.0 +0x8ca4e void* gsi::SerialArgs::read<void*>(tl::Heap&) [??:?]
/lib64/libruby.so.2.0 +0x164593 rb_catch [??:?]
/lib64/libruby.so.2.0 +0x174034 rb_each [??:?]
/lib64/libruby.so.2.0 +0x163f4e rb_iterate [??:?]
/lib64/libruby.so.2.0 +0x1640bb rb_block_call [??:?]
/lib64/libruby.so.2.0 +0x164593 rb_catch [??:?]
/lib64/libruby.so.2.0 +0x174034 rb_each [??:?]
/lib64/libruby.so.2.0 +0x163f4e rb_iterate [??:?]

==> klay-tst4-0.29.1.log <==
RUBY_VERSION=2.0.0
indices: [0, 1, 2, 3]
indices: [0, 1, 2, 3]   do_1: true
indices: [0, 1, 2, 3]   do_3: true
test index 0:
... not_empty_0.class: NilClass
test index 3:
... not_empty_3.class: RBA::Cell
test index 1:
ERROR: Signal number: 11
Address: 0x18
Program Version: KLayout 0.29.1 (2024-05-04 rf95aef8)

Backtrace:
/usr/lib64/klayout/libklayout_lay.so.0 +0x29778e lay::enable_signal_handler_gui(bool) [??:?]
/lib64/libpthread.so.0 +0xf630 __restore_rt [sigaction.c:?]
/lib64/libruby.so.2.0 +0x16d958 rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x171a20 rb_yield [??:?]
/lib64/libruby.so.2.0 +0x5fa5e rb_enum_values_pack [??:?]
/lib64/libruby.so.2.0 +0x171c80 rb_yield [??:?]
/lib64/libruby.so.2.0 +0x69d0b rb_protect [??:?]
/usr/lib64/klayout/libklayout_rba.so.0 +0x45a8b void std::vector<std::pair<std::string, gsi::MethodBase const*>, std::allocator<std::pair<std::string, gsi::MethodBase const*> > >::_M_emplace_back_aux<std::pair<std::string, gsi::MethodBase const*> >(std::pair<std::string, gsi::MethodBase const*>&&) [??:?]
/usr/lib64/klayout/libklayout_rba.so.0 +0x2567a rba::RubyInterpreter::add_package_location(std::string const&) [??:?]
/lib64/libruby.so.2.0 +0x164593 rb_catch [??:?]
/lib64/libruby.so.2.0 +0x174034 rb_each [??:?]
/lib64/libruby.so.2.0 +0x163f4e rb_iterate [??:?]
/lib64/libruby.so.2.0 +0x1640bb rb_block_call [??:?]
/lib64/libruby.so.2.0 +0x164593 rb_catch [??:?]
/lib64/libruby.so.2.0 +0x174034 rb_each [??:?]
/lib64/libruby.so.2.0 +0x163f4e rb_iterate [??:?]
```
[klay-empty-crash.tgz](https://github.com/user-attachments/files/15817872/klay-empty-crash.tgz)

klayoutmatthias, 2024-06-13:
Hi @sto6,

Maybe I should have said, that the "each_cell.find(...)" idiom is based on the Enumerator binding of the iterators inside KLayout's Ruby API code. This feature was not stable in the past. I am pretty sure this is why you see this crash.

For backward compatibility with older versions, I'd recommend to rewrite the loop to

```
def layer_is_not_empty(layout, layer)
  result = false
  layout.each_cell do |c|
    if !c.shapes(layer).is_empty?
      result = true
      break
    end
  end
  result
end

# use it like this:
not_empty_0 = layer_is_empty(ly, 0) 
```

Matthias

sto6, 2024-06-13:
Thank you. Does same (unstable) likely apply also to the begin_shapes_rec iterators?

klayoutmatthias, 2024-06-14:
No, that is perfectly normal code. It has to work.

The stack traces are weird. You are sure there is no mixup of libraries from different versions?

Matthias

klayoutmatthias, 2024-06-15:
I just tried with a CentOS 7 Docker instance and the 0.29.2 RPM from here: 

Two observations:

* The original version of your script crashes with the strange stack trace as you describe it. I think this is because of the old Ruby version, KLayout's Enumerator binding is not fully compatible with. I don't know if it is really worth debugging the issue. I don't see it on recent Ruby versions.
* A modified version of the script using the "layer_not_empty" function works. I have copied the modified version below.

Matthias

```
#
# run: klayout -b -r klay-tst4.rb
#
# test.oas input layout-file hardcoded, has indices 0...3, only 3 (46/0) is non-empty:
#   0 Via3 (40/0)  name='Via3'  layer='40'  type='0'
#   1 my.14.223 (14/223)  name='my.14.223'  layer='14'  type='223'
#   2 my12.222 (12/222)  name='my12.222'  layer='12'  type='222'
#   3 46/0  name=''  layer='46'  type='0'
#
# In klayout < 0.29.0, only layer 46/0 (as index 0) will exist.
#
# Script will test for not_empty layers of this sequence of indices:
#  >= 0.29.0: 0 3*  1** 0
#  <  0.29.0: 0*        0**
# *: only non-empty layer (46/0) in the test.oas
# **: crash during (or soon after) 'ly.each_cell.find'
#
# After the 1st 'ly.each_cell.find' that DOES find a non-empty layer,
# klayout is very prone to crash: another ly.each_cell.find (whether empty or not)
# would crash it, and an exit also will crash.
#
puts "RUBY_VERSION=#{RUBY_VERSION}"

include RBA
ly = Layout.new
ly.read("test.oas")    # versus: ly.read("test.gds")
puts "indices: #{ly.layer_indices.to_s}"

do_1 = ly.layer_indices.member?(1)
puts "indices: #{ly.layer_indices.to_s}   do_1: #{do_1.to_s}"
do_3 = ly.layer_indices.member?(3)
puts "indices: #{ly.layer_indices.to_s}   do_3: #{do_3.to_s}"

def layer_is_not_empty(layout, layer)
  result = false
  layout.each_cell do |c|
    if !c.shapes(layer).is_empty?
      result = true
      break
    end
  end
  result
end

STDOUT.puts "test index 0:"
STDOUT.flush
STDERR.flush
not_empty_0 = layer_is_not_empty(ly, 0)  # <-- known empty
STDOUT.puts "... not_empty_0.class: #{not_empty_0.class}"
STDOUT.flush

if do_3
  STDOUT.puts "test index 3:"
  STDOUT.flush
  STDERR.flush
  not_empty_3 = layer_is_not_empty(ly, 3)  # <-- known NOT empty
  STDOUT.puts "... not_empty_3.class: #{not_empty_3.class}"
  STDOUT.flush
end

if do_1
  STDOUT.puts "test index 1:"
  STDOUT.flush
  STDERR.flush
  not_empty_1 = layer_is_not_empty(ly, 1)  # <-- known empty
  STDOUT.puts "... not_empty_1.class: #{not_empty_1.class}"
  STDOUT.flush
end

STDOUT.puts "test index 0 (again):"
STDOUT.flush
STDERR.flush
not_empty_0 = layer_is_not_empty(ly, 0)  # <-- known empty
STDOUT.puts "... not_empty_0.class: #{not_empty_0.class}"
STDOUT.flush

STDOUT.puts "Exiting..."
STDOUT.flush

# emacs syntax-mode:
# Local Variables:
# mode:ruby
# End:
```

klayoutmatthias, 2024-06-15:
Plus another data point: I ran the same test with CentOS 8 and the RPM from here: https://www.klayout.org/downloads/CentOS_8/klayout-0.29.2-0.x86_64.rpm

This combination comes with Ruby 2.5.9-p229. The original version of your script works without problems.

Matthias

klayoutmatthias, 2024-06-15:
Sorry, closed by mistake. The ticket is still open.

sto6, 2024-06-17:
Sorry, did not mean to imply I had _confirmed_ an issue with begin_shapes_rec.
Just trying to predict or infer what else is _likely_ to have same issue (more general scope than layer-empty) when reading the class docs.
Anything marked `[iter]`, if used with .find, .any? etc. may have the issue?
Like in class:Cell, method:each_inst.
In RecursiveShapeIterator, method: each (if used with .find, any? ...); but looping using .at_end? and .next would work.

klayoutmatthias, 2024-06-17:
No worries. Maybe I need to explain a little more.

When you find `[iter]` as "return type" in the documentation, it means that the method can take a block and will yield the block with every value it represents.

Like

```
ly = ... some Layout ...
ly.each_cell do |c|
  ... called for each cell in c ...
end
```

This is the basic use case and it should always work. 

A little side note: I noticed that because of the C++/Ruby transition in the iterator implementation, a "return" from inside the block acts like "break" and will not actually return. I will fix this in the next minor released.

Anyway, there is a second feature. If you use such a method without a block, it will return an "Enumerator" object - this is a Ruby feature and basically means an iterator. The enumerator is a pretty powerful concept in Ruby and binds to many methods, like "collect":

```
ly = ... some Layout ...
all_cell_names = ly.each_cell.collect { |c| c.name }
```

This is the use case that seems not to work with Ruby 2.0.0. I have not debugged the problem, but I assume it is hidden in Ruby's memory management. That fact that it works with all other Ruby versions I have in place tells me that the implementation can't be that wrong. Debugging Ruby management problems is painful and I'd rather count on Ruby 2.0.0 vanishing soon with CentOS 7.

Regarding `RecursiveShapeIterator` there is another pitfall as highlighted here: https://github.com/KLayout/klayout/issues/1742. I can act as an enumerator (through "each"), but that is a pseudo-iterator. It will just deliver "self". So stashing these values will give you a list of the object many times.

But using "each" to iterate basically is fine. This is equivalent:

```
iter = ... some RecursiveShapeIterator ...
while !iter.at_end?
  ... do something with iter ..
  iter.next
end
```

and

```
iter = ... some RecursiveShapeIterator ...
iter.each do |i|
  ... do something with i (which is the same than iter) ...
end
```

In better Ruby versions, you can also use ".find" or ".collect":

```
iter = ... some RecursiveShapeIterator ...
# caution: don't use this on Ruby 2.0.0
all_shapes = iter.each.collect { |i| i.shape }
```

(this is a somewhat contrieved example as without transformation, the shape is somewhat useless).

Matthias

klayoutmatthias, 2024-07-27:
Can I close this issue?