KLayout/klayout #1742: Segmentation Fault

frankxu-omni, 2024-06-14:
[image]
line14 cause Segmentation Fault
[demo_seg_fault.zip](https://github.com/user-attachments/files/15841761/demo_seg_fault.zip)

klayoutmatthias, 2024-06-15:
Hi @frankxu-omni,

thanks for reporting that problem and the nice test case.

Problem is here the following:

*  `begin_shapes_rec(..)` delivers a `RecursiveShapeIterator`. 
* This object can be iterated, but upon iteration, it only delivers 'self'. This meant by the description of "each" (the implicit iterator) here: https://www.klayout.de/doc-qt5/code/class_RecursiveShapeIterator.html#method22.
* This means, in the list you get two references to `self` and after iteration is done, the iterator is in "at_end" state - i.e. invalid.
* So eventually in your list you have two references to invalid shape iterators.
* Apparently "inspect" does something on the object which - in invalid state - causes the segfault. I don't know if one could call that a bug as this is somewhat of a border case. But I can try to debug the issue and prevent the crash in that case.

Anyway, the solution is to store deep copies of the iterator:

```
l = [ i.dup() for i in top.begin_shapes_rec(1) ]
foo = l[0]
inspect(foo)
```

I'd like to point out that `RecursiveShapeIterator` is not a lightweight object. If you iterate over many shapes, you would create many copies of them. It may be more efficient to remember the attribute values you are interested in. 

Matthias

frankxu-omni, 2024-06-17:
Hi Matthias, 
Got it. Thanks for the effort.
-Frank

closed 2024-06-20 by frankxu-omni