KLayout/klayout #1743: strmxor shows not difference, klayout xor shows 85, dbu 99.999 vs 1000

stefanottili, 2024-06-15:
strmxor a.oas b.oas xor.oas shows not differences
klayout a.oas -s b.oas, xor with 500 tile, healing shows 85 differences.

a.oas 99.999 dbu vs b.oas 1000 dbu, only 10 polygons.

[test_klayout_scaled_xor.tgz](https://github.com/user-attachments/files/15855572/test_klayout_scaled_xor.tgz)

klayoutmatthias, 2024-06-17:
Okay, there is a difference between strmxor and KLayout XOR when it comes to different DBUs:

* strmxor will use the minimum of both and scale the other layout accordingly.
* KLayout XOR will use the LCM. This is meant for DBU pairs where one is a multiple of the other.

In the test case, the LCM of 0.0100000016391 and 0.001 is, ahem, 83265.1532405.

Maybe it is better to take the minimum for the KLayout XOR too.

stefanottili, 2024-06-17:
Ah, ok. There are only a handful of DBU’s that make sense, I’l try to avoid writing not rounded ones.

closed 2024-07-01 by klayoutmatthias