KLayout/klayout #1745: MacBook Air M3 chip Sonoma 14.5

ChihYuLee, 2024-06-17:
Hi, I am wondering what software I should use for this hardware setup? I have tried those for Anaconda user Sonoma 14.1, all of them quit unexpectedly. Is there something wrong with the new Mac?

stefanottili, 2024-06-17:
a) you need the anaconda dependencies installed and
b) the executables are x86, so you need Rosetta.

On my M1, I’m using homebrew to get the dependencies, git clone klayout and use build4mac to compile it to run on arm64.

Kazzz-S, 2024-06-17:
Hi @stefanottili,

Thanks for your comments :-)

----

Hi @ChihYuLee,

A “heavy” download is like **_HW_-klayout-0.29.2-macOS-Sonoma-1-qt5MP-RsysPhb311.dmg**.

My colleague confirmed he could use the `HW*.dmg` package on his M2 Sonoma with `Rosetta`.
I think the `ST*.dmg` might also work with `Rosetta`.

If you try to build a native executable from the source, refer to [this document](https://github.com/KLayout/klayout/tree/master/macbuild).
However, please note that the above document needs to be updated.
More precisely, **6C. Fully Homebrew-flavored build...** and **6D. Partially Homebrew-flavored build...** are now OK.

Kazzz-S

ChihYuLee, 2024-06-17:
Thank you, the HW works, I didn't know why my Anaconda doesn't work.