KLayout/klayout #1746: Python API assigning "each_shape" iterator  to a variable causing rendering issue

s910324, 2024-06-17:
Operation sysrtem : Windows 11 x64
Version tested : 0.28.15 ~0.29.2 (portable)

Assigning each_shape iter to a variable `it = top.each_shape(layer)`  often causes `Shapes` / `Cell` in corrosponding view disappear.

* it seems like only affect the UI, the data was not affected.

https://github.com/KLayout/klayout/assets/1561043/92ef1de5-e315-4c05-ac25-13c8e81fc7ab

Minimal python test case :
``` python
import pya
layout   = pya.Application.instance().main_window().current_view().active_cellview().layout()
top      = layout.cell("TOP")
layer    = layout.layer(1, 0)
it       = top.each_shape(layer) # this will causes issue

for s in  top.each_shape(layer) : # not assigning to varible will not cause issus
  print(s)
```

I've also done similar operation using ruby, which will not have the same issue
Minimal ruby test case :
```ruby
layout   = RBA::Application::instance::main_window.current_view.active_cellview.layout
top      = layout.cell("TOP")
layer    = layout.layer(1, 0)
it       = top.each_shape(layer)
it.each{ |s| puts s}
```

klayoutmatthias, 2024-06-17:
Hi @s910324

That is intended. While you hold an iterator in a variable, it will lock the layout (technically: put it into "under_construction" mode).
While a layout is under construction it does not rebuild its search trees or bounding box caches. This causes the effects you observe.

The reason behind this decision is to prevent internal updates to interfere with the iterator. For example, rebuilding the search trees will change the order of the shapes. So while iterating, such an update must not happen - for example as an effect of another operation. This is what "under construction" does.

I'd advise against keeping iterators for longer. Iterators are meant to be short-living objects needed during a loop only.

Matthias

s910324, 2024-06-18:
Hi Matthias, 

Thanks for the explanation, and agree, that this type of operation indeed needs to be used and processed right away, storing the iterator is probably not a good idea.

closed 2024-06-18 by s910324